platformio/platformio-vscode-ide #1043: Negative value on unsigned short variable

giorgakos, 2019-10-02:
Unsigned short variable with value equal to -1 produces no warnings or errors

ivankravets, 2019-10-11:
Could you provide more details?

giorgakos, 2019-10-11:
Sure, 

unsigned short variableName = -1;

compiles just fine while using C,
via the C, C++ extension from microsoft.

ivankravets, 2019-10-11:
Please move to https://github.com/microsoft/vscode-cpptools/issues

giorgakos, 2019-10-11:
i have already contacted microsoft about this issue, and they closed it as they say that this one is not their issue

ivankravets, 2019-10-11:
Could you point me to their issue?

giorgakos, 2019-10-11:
https://github.com/microsoft/vscode/issues/81794

ivankravets, 2019-10-11:
You need to enable this error for GCC via build flags. See http://docs.platformio.org/en/latest/projectconf/section_env_build.html#build-flags

See `-Wall` and `-Werror`.

closed 2019-10-11 by ivankravets