platformio/platformio-vscode-ide #1052: seach by ID #

prabbit237, 2019-10-05:


prabbit237, 2019-10-05:
Ok, I've got old and poorly commented in an .ini file. I wish to look at what the libs are to try and clean up the ,ini. They are in like:
        lib_deps = 
             13
             256
             9
I want to clean it up some and label them better. How in Visual Code use the Platformio search to say what each are? Search for  "13" does no good. ID# 13 also no go. I CAN go to a term window and use "pio lib search --id 13" and find it but that's not as easy as in the PlatformIO Home Page..

Also can the "pio lib search <something> "not load up three or four screens before pausing? How about paging it as a PAGE that fits on one screen?

ivankravets, 2019-10-11:
You can do that with `id:13`  PIO Home

closed 2019-10-11 by ivankravets