platformio/platformio-vscode-ide #1063: Installation Manager

majay123, 2019-10-09:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.38.1
PIO IDE: v1.8.3
System: Linux, 3.2.0-23-generic, x64

%23 Exception
```
Error: PIP Core: /usr/bin/python2: No module named pip

	at o.runCommand (/usr/mcd/.vscode-server/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:32129)
	at ChildProcess.h (/usr/mcd/.vscode-server/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6057)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe._handle.close (net.js:606:12)
```

ivankravets, 2019-10-11:
Do you have `python3 --version` in a system?

ivankravets, 2019-10-11:
Could you try "Solution" from https://github.com/platformio/platformio-vscode-ide/issues/959 ?
You will need to install PlatformIO IDE for VSCode 1.9.0-beta.

Does it work now?

closed 2019-10-18 by ivankravets