platformio/platformio-vscode-ide #1067: Start PIO Home Server

zen29d, 2019-10-09:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.38.1
PIO IDE: v1.8.3
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at S.Promise (/Users/zen/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:17372)
	at ChildProcess.h (/Users/zen/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe.Socket._destroy._handle.close (net.js:606:12)
```

ivankravets, 2019-10-10:
Could you provide an output from these commands?
```
python -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
python3 -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
```

zen29d, 2019-10-10:
System: macOS Catalina
PlatformIO installation using python3
Previous error was generated by installing form VSCode

> python -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))”

sys.version_info(major=3, minor=7, micro=3, releaselevel='final', serial=0) <sha1 HASH object @ 0x1018b6a58>

ivankravets, 2019-10-11:
Could you try "Solution" from https://github.com/platformio/platformio-vscode-ide/issues/959 ?
You will need to install PlatformIO IDE for VSCode 1.9.0-beta.

Does it work now?

ivankravets, 2019-10-11:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/959

closed 2019-10-11 by ivankravets

zen29d, 2019-10-11:
Thanks, I solved by installing paltformio-core and linking with VSCode

ivankravets, 2019-10-11:
What is your?
```
~/.platformio/penv/bin/python --version
```

zen29d, 2019-10-11:
Python 3.7.3

ivankravets, 2019-10-11:
How did you solve it?

zen29d, 2019-10-11:
Second time when I'm trying to install it popping another error (like: python2.7 error), so I just install platformio-core uisng pip (python3.7) and VSCode's platformio extension not able to installed core its but setting is changeable, so i change "platformio-ide.useBuiltinPIOCore": false,
"platformio-ide.customPATH": "/Users/zen/Applications/.platformio/penv/bin", 
# Successfully able compile arduino code
# [bug/issue: when trying to install from extension its always choose Python2 not the default python of system: mine was Python3.7 ]

ivankravets, 2019-10-11:
1. Please revert back useBUiltIn PIO Core to `true`
2. Remove customPATH
3. Install PlatformIO IDE 1.9.0 from VSIX. See https://github.com/platformio/platformio-vscode-ide/issues/959

Restart VSCode.

Does it work now and use your Python 3?

zen29d, 2019-10-11:
The below one is result of installing again: Conclude - I installed Conda_Python3.7, so error is expected: my default python is Python3.7_Conda but platformio extension is failing use python3, but it work with previous method. 
>Error: Conda Virtualenv: WARNING conda.core.package_cache_data:_make_single_record(361): >Encountered corrupt package tarball at /Users/zen/Applications/miniconda3/pkgs/python-2.7.16-h97142e2_7.conda. Conda has removed it, but you need to re-run conda to download it again.

>[Errno 30] Read-only file system: '/.cph_tmpsj4g973_'

	at o.runCommand (/Users/zen/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:29363)
	at ChildProcess.h (/Users/zen/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe._handle.close (net.js:606:12)

ivankravets, 2019-10-11:
Ah, this is a broken installation. Please destroy this conda and use official Python.

zen29d, 2019-10-11:
The Problem is not with Conda, I have differnt system with conda_Python2, and its work fine with conda python2. But I need to work with python3. And the working platformIO-core with Conda_python3 I uses without any problem. The issue is VSIX not able to use python3, it always try to call python2.

ivankravets, 2019-10-11:
It looks for Python 3 firstly. That is why I asked you to try it. It seems that conda's python 3 installation is broken and can pass basic tests. See

- https://github.com/platformio/platformio-node-helpers/blob/master/src/misc.js#L155
- https://github.com/platformio/platformio-node-helpers/blob/master/src/misc.js#L185