platformio/platformio-vscode-ide #1068: Start PIO Home Server

Soofiyan, 2019-10-09:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.38.1
PIO IDE: v1.8.3
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at S.Promise (/Users/soofiyanatar/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:17372)
	at ChildProcess.h (/Users/soofiyanatar/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe.Socket._destroy._handle.close (net.js:606:12)
```

ivankravets, 2019-10-10:
Could you provide an output from these commands?
```
python -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
python3 -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
```

Soofiyan, 2019-10-11:
(sys.version_info(major=2, minor=7, micro=16, releaselevel='final', serial=0), <sha1 HASH object @ 0x109a6a788>) 

sys.version_info(major=3, minor=7, micro=4, releaselevel='final', serial=0) <sha1 HASH object @ 0x10c8d0ae0>

ivankravets, 2019-10-11:
Could you try "Solution" from https://github.com/platformio/platformio-vscode-ide/issues/959 ?
You will need to install PlatformIO IDE for VSCode 1.9.0-beta.

Does it work now?

ivankravets, 2019-10-11:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/959

ivankravets, 2019-10-11:
Please provide an output from
```
~/.platformio/penv/bin/python --version
```

Soofiyan, 2019-10-11:
Python 3.7.4

> 
> Please provide an output from
> 
> ~/.platformio/penv/bin/python --version

ivankravets, 2019-10-11:
Do you see any errors when running ?
```
~/.platformio/penv/bin/python -m platformio home
```

Soofiyan, 2019-10-11:
/Users/soofiyanatar/.platformio/penv/bin/python: No module named platformio

> 
> Do you see any errors when running ?
> 
> ~/.platformio/penv/bin/python -m platformio home

ivankravets, 2019-10-11:
Hm... something wrong with your installation. Please remove `~/.platformio/penv` folder and restart VSCode.

Soofiyan, 2019-10-11:
zsh: abort      ~/.platformio/penv/bin/python -m platformio home

This error was seen for ~/.platformio/penv/bin/python -m platformio home command

> 
> Hm... something wrong with your installation. Please remove ~/.platformio/penv folder and restart VSCode.

ivankravets, 2019-10-11:
Aha, this is internal OS issue. Could you open macOS `Console.app`, clear screen and try to repeat again:
```
~/.platformio/penv/bin/python -m platformio home
```

Please post exception here.

Soofiyan, 2019-10-11:
How to run on console.app?
Please help

ivankravets, 2019-10-11:
'/Applicaitons/Utilities/Console.app'

Soofiyan, 2019-10-11:
Yaa I know console but how to run command on that app

Soofiyan, 2019-10-11:
please help

ivankravets, 2019-10-11:
Just keep this app open. When any app crashes in the system, you will see errors here.

Soofiyan, 2019-10-11:
ok thanks

xtianpaiva, 2019-10-11:
> Aha, this is internal OS issue. Could you open macOS `Console.app`, clear screen and try to repeat again:
> 
> ```
> ~/.platformio/penv/bin/python -m platformio home
> ```
> 
> Please post exception here.

```
Process:               python [2218]
Path:                  /Users/USER/*/python
Identifier:            python
Version:               125
Code Type:             X86-64 (Native)
Parent Process:        zsh [2210]
Responsible:           Terminal [2208]
User ID:               501

Date/Time:             2019-10-11 15:38:12.717 -0300
OS Version:            Mac OS X 10.15 (19A583)
Report Version:        12
Bridge OS Version:     3.0 (14Y906)
Anonymous UUID:        E4768B04-A02C-7850-94BD-270464FAE5A5

Time Awake Since Boot: 3500 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
/usr/lib/libcrypto.dylib
abort() called
Invalid dylib load. Clients should not load the unversioned libcrypto dylib as it does not have a stable ABI.

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff65c4647a __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff65d03707 pthread_kill + 384
2   libsystem_c.dylib             	0x00007fff65bcea08 abort + 120
3   libcrypto.dylib               	0x00007fff63558804 __report_load + 352
4   dyld                          	0x0000000111f0b635 ImageLoaderMachO::doModInitFunctions(ImageLoader::LinkContext const&) + 539
5   dyld                          	0x0000000111f0ba5a ImageLoaderMachO::doInitialization(ImageLoader::LinkContext const&) + 40
6   dyld                          	0x0000000111f06933 ImageLoader::recursiveInitialization(ImageLoader::LinkContext const&, unsigned int, char const*, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) + 493
7   dyld                          	0x0000000111f0569c ImageLoader::processInitializers(ImageLoader::LinkContext const&, unsigned int, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) + 188
8   dyld                          	0x0000000111f0573c ImageLoader::runInitializers(ImageLoader::LinkContext const&, ImageLoader::InitializerTimingList&) + 82
9   dyld                          	0x0000000111ef7d87 dyld::runInitializers(ImageLoader*) + 82
10  dyld                          	0x0000000111f01767 dlopen_internal + 609
11  libdyld.dylib                 	0x00007fff65ae7f6f dlopen + 171
12  _ctypes.so                    	0x0000000108382208 0x108379000 + 37384
13  org.python.python             	0x00007fff3972241c PyEval_EvalFrameEx + 20654
14  org.python.python             	0x00007fff3971cd5e PyEval_EvalCodeEx + 534
15  org.python.python             	0x00007fff396c3c85 0x7fff3969c000 + 162949
16  org.python.python             	0x00007fff396a6319 PyObject_Call + 97
17  org.python.python             	0x00007fff396b0afe 0x7fff3969c000 + 84734
18  org.python.python             	0x00007fff396a6319 PyObject_Call + 97
19  org.python.python             	0x00007fff396ec605 0x7fff3969c000 + 329221
20  org.python.python             	0x00007fff396e9807 0x7fff3969c000 + 317447
21  org.python.python             	0x00007fff396a6319 PyObject_Call + 97
22  org.python.python             	0x00007fff3972240a PyEval_EvalFrameEx + 20636
23  org.python.python             	0x00007fff3971cd5e PyEval_EvalCodeEx + 534
24  org.python.python             	0x00007fff3971cb42 PyEval_EvalCode + 32
25  org.python.python             	0x00007fff397342ae PyImport_ExecCodeModuleEx + 212
26  org.python.python             	0x00007fff39737033 0x7fff3969c000 + 634931
27  org.python.python             	0x00007fff39736aff 0x7fff3969c000 + 633599
28  org.python.python             	0x00007fff397366d7 0x7fff3969c000 + 632535
29  org.python.python             	0x00007fff39735720 PyImport_ImportModuleLevel + 656
[... 451 lines cut ...]
 
                                VIRTUAL   REGION 
REGION TYPE                        SIZE    COUNT (non-coalesced) 
===========                     =======  ======= 
Activity Tracing                   256K        1 
Dispatch continuations            8192K        1 
Kernel Alloc Once                    8K        1 
MALLOC                            41.3M       37 
MALLOC guard page                   16K        4 
STACK GUARD                       56.0M        4 
Stack                             13.0M        4 
VM_ALLOCATE                       11.5M       47 
__DATA                            6917K      164 
__DATA_CONST                       152K       33 
__LINKEDIT                       349.0M       35 
__OBJC_RO                         31.8M        1 
__OBJC_RW                         1764K        2 
__TEXT                            71.0M      162 
__UNICODE                          564K        1 
mapped file                         28K        1 
shared memory                       44K        5 
===========                     =======  ======= 
TOTAL                            591.2M      503 

```

ivankravets, 2019-10-13:
Could you try the next?

1. Reinstall Python from the source which you used before:
    a) if it's `brew`, then `brew uninstall python`, `brew install python`
    b) if it's Python from Apple Store, just uninstall and install again
    c) if it's official Python from Python Org, just download the latest Python 3.7 and re-install it.

2. Open system terminal and type `rm -rf ~/.platformio/penv`
3. Restart VScode.

Does it work now?

xtianpaiva, 2019-10-13:
[image]

[image]

`rm -rf ~/.platformio/penv` did the trick for me, no Python reinstall was necessary (recently upgraded to macOS Catalina, Python 2.7.16 is included).

ivankravets, 2019-10-13:
Happy coding with PlatformIO!

closed 2019-10-13 by ivankravets