platformio/platformio-vscode-ide #1069: Start PIO Home Server

tamusiunas, 2019-10-09:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.0
PIO IDE: v1.8.3
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at S.Promise (/Users/fabricio/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:17372)
	at ChildProcess.h (/Users/fabricio/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.EventEmitter.emit (domain.js:442:20)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-10-10:
Could you provide an output from these commands?
```
python -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
python3 -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
```

tamusiunas, 2019-10-10:
fabricio@Fabricios-MBP LaunchDaemons % python3 -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
sys.version_info(major=3, minor=7, micro=4, releaselevel='final', serial=0) <sha1 HASH object @ 0x10b869b10>
fabricio@Fabricios-MBP LaunchDaemons % python -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
(sys.version_info(major=2, minor=7, micro=16, releaselevel='final', serial=0), <sha1 HASH object @ 0x10c9ee7d8>)

I'm trying to run it on macOS Catalina with Python 3 installed from Homebrew.

ivankravets, 2019-10-11:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/959

closed 2019-10-11 by ivankravets