platformio/platformio-vscode-ide #1070: Start PIO Home Server

Soofiyan, 2019-10-10:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.0
PIO IDE: v1.8.3
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at S.Promise (/Users/soofiyanatar/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:17372)
	at ChildProcess.h (/Users/soofiyanatar/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-10-10:
Could you provide an output from these commands?
```
python -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
python3 -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
```

Soofiyan, 2019-10-11:
(sys.version_info(major=2, minor=7, micro=16, releaselevel='final', serial=0), <sha1 HASH object @ 0x109a6a788>)

sys.version_info(major=3, minor=7, micro=4, releaselevel='final', serial=0) <sha1 HASH object @ 0x10c8d0ae0>

ivankravets, 2019-10-11:
Could you try "Solution" from https://github.com/platformio/platformio-vscode-ide/issues/959 ?
You will need to install PlatformIO IDE for VSCode 1.9.0-beta.

Does it work now?

ivankravets, 2019-10-11:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/959

closed 2019-10-11 by ivankravets