platformio/platformio-vscode-ide #1071: Start PIO Home Server

0x6e3078, 2019-10-10:
%23 Description of problem
After Update of VSCode, this issue happens. 

```
platformio upgrade
You're up-to-date!
PlatformIO 4.0.3 is currently the newest version available.
```

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.0
PIO IDE: v1.8.3
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at S.Promise (/Users/sgu/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:17372)
	at ChildProcess.h (/Users/sgu/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe.Socket._destroy._handle.close (net.js:606:12)
```

ivankravets, 2019-10-10:
Could you provide an output from these commands?
```
python -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
python3 -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
```

ivankravets, 2019-10-11:
Could you try "Solution" from https://github.com/platformio/platformio-vscode-ide/issues/959 ?
You will need to install PlatformIO IDE for VSCode 1.9.0-beta.

Does it work now?

ivankravets, 2019-10-11:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/959

closed 2019-10-11 by ivankravets