platformio/platformio-vscode-ide #1075: Start PIO Home Server

lewisxhe, 2019-10-10:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.0
PIO IDE: v1.8.3
System: Darwin, 18.7.0, x64

%23 Exception
```
Error: Error: Please read http://bit.ly/package-manager-ioerror
Error: Please read http://bit.ly/package-manager-ioerror
Error: Could not install 'contrib-pysite' with version requirements '~2.37.190418' for your system 'darwin_x86_64'.

Please try this solution -> http://bit.ly/faq-package-manager

	at S.Promise (/Users/lewis/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:17372)
	at ChildProcess.h (/Users/lewis/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe.Socket._destroy._handle.close (net.js:606:12)
```

ivankravets, 2019-10-10:
Could you provide an output from these commands?
```
python -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
python3 -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
```

ivankravets, 2019-10-11:
Please try "Solution" from https://github.com/platformio/platformio-vscode-ide/issues/959

closed 2019-10-11 by ivankravets