platformio/platformio-vscode-ide #1076: Start PIO Home Server

blagdjinn, 2019-10-10:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.0
PIO IDE: v1.8.1
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at S.Promise (/Users/blagdjinn/.vscode/extensions/platformio.platformio-ide-1.8.1/node_modules/platformio-node-helpers/dist/index.js:1:17211)
	at ChildProcess.h (/Users/blagdjinn/.vscode/extensions/platformio.platformio-ide-1.8.1/node_modules/platformio-node-helpers/dist/index.js:1:6019)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

blagdjinn, 2019-10-10:
blagdjinn@MBP-de-Gilles ~ % python -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
(sys.version_info(major=2, minor=7, micro=16, releaselevel='final', serial=0), <sha1 HASH object @ 0x10777a7d8>)

blagdjinn@MBP-de-Gilles ~ % python3 -c "import hashlib,sys; print(sys.version_info, hashlib.sha1(b'a'))"
sys.version_info(major=3, minor=7, micro=3, releaselevel='final', serial=0) <sha1 HASH object @ 0x102c1e670>

ivankravets, 2019-10-10:
@blagdjinn did you install Python 3 manually?

Please provide 
```
python3 -c "import hashlib;print(hashlib)"
```

blagdjinn, 2019-10-10:
python3 -c "import hashlib;print(hashlib)"
<module 'hashlib' from '/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/hashlib.py'>

This is a clean install of Catalina OSX. And a new install of platform with vscode.

Python wasn't install manually. It must have been install by default..

ivankravets, 2019-10-10:
Thanks! I work on the fix.

IAmOrion, 2019-10-11:
+1 - I'm also having this issue.  Clean MacOS Catalina install

`(sys.version_info(major=2, minor=7, micro=16, releaselevel='final', serial=0), <sha1 HASH object @ 0x10363b7d8>)`

`sys.version_info(major=3, minor=7, micro=3, releaselevel='final', serial=0) <sha1 HASH object @ 0x106b366c0>`

`<module 'hashlib' from '/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/hashlib.py'>`

ivankravets, 2019-10-11:
Could you try "Solution" from https://github.com/platformio/platformio-vscode-ide/issues/959 ?
You will need to install PlatformIO IDE for VSCode 1.9.0-beta.

Does it work now?

ivankravets, 2019-10-11:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/959

closed 2019-10-11 by ivankravets