platformio/platformio-vscode-ide #1079: Start PIO Home Server

lewisxhe, 2019-10-11:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.1
PIO IDE: v1.9.0-beta.2
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: Error: Please read http://bit.ly/package-manager-ioerror
Error: Please read http://bit.ly/package-manager-ioerror
Error: Could not install 'contrib-pysite' with version requirements '~2.37.190418' for your system 'darwin_x86_64'.

Please try this solution -> http://bit.ly/faq-package-manager

	at /Users/lewis/.vscode/extensions/platformio.platformio-ide-1.9.0-beta.2/node_modules/platformio-node-helpers/dist/index.js:1:17383
	at ChildProcess.h (/Users/lewis/.vscode/extensions/platformio.platformio-ide-1.9.0-beta.2/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe._handle.close (net.js:606:12)
```

lewisxhe, 2019-10-11:
I suspect that it has a relationship with the Python version. I have two macs, the same is updated for mac Catalina, one is python3.6.5, one is python3.7.2, version 3.6.5 is running normally, and the 3.7 version is always stuck in loading. ...interface

ivankravets, 2019-10-11:
> Error: Error: Please read http://bit.ly/package-manager-ioerror

It seems like an internal issue of VSCode terminal. Try to open macOS native terminal and type
```
~/.platformio/penv/bin/platformio home
```

lewisxhe, 2019-10-11:
I deleted python3.7.2 through homebrew, and then installed the latest python3.7.5, still can't solve it. Finally, I am mad, I directly delete python3, reinstall pio using python2 that comes with mac, and solve it smoothly. @ivankravets

closed 2019-10-11 by lewisxhe