platformio/platformio-vscode-ide #1080: Start PIO Home Server

deamens, 2019-10-11:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.1
PIO IDE: v1.8.3
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at /Users/dougeamens/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:17372
	at ChildProcess.h (/Users/dougeamens/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe._handle.close (net.js:606:12)
```

ivankravets, 2019-10-11:
Could you try solution from https://github.com/platformio/platformio-vscode-ide/issues/959 ?

Did it help you?

closed 2019-10-11 by ivankravets

deamens, 2019-10-11:
Sorry i didnt see that fix..i just tried it and its fixed the
problem...thank you