platformio/platformio-vscode-ide #1081: Start PIO Home Server

nathanribeiroo, 2019-10-11:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.0
PIO IDE: v1.8.3
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at S.Promise (/Users/nathanribeiro/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:17372)
	at ChildProcess.h (/Users/nathanribeiro/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-10-11:
Could you try solution from https://github.com/platformio/platformio-vscode-ide/issues/959 ?

Did it help you?

nathanribeiroo, 2019-10-11:
Thanks Ivan, that solved

nathanribeiroo, 2019-10-11:
@ivankravets Thanks Ivan, that solved

closed 2019-10-11 by nathanribeiroo

ivankravets, 2019-10-11:
Great! We will release 1.9.0 soon.