platformio/platformio-vscode-ide #1084: Installation Manager

jlopezt, 2019-10-11:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.1
PIO IDE: v1.9.0
System: Windows_NT, 10.0.17763, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\jlopezt\.platformio\.cache\tmp-43287b5n17Z65lk4\virtualenv-16.7.5\virtualenv.py", line 2632, in <module>
    main()
  File "C:\Users\jlopezt\.platformio\.cache\tmp-43287b5n17Z65lk4\virtualenv-16.7.5\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\Users\jlopezt\.platformio\.cache\tmp-43287b5n17Z65lk4\virtualenv-16.7.5\virtualenv.py", line 1173, in create_environment
    install_wheel(to_install, py_executable, search_dirs, download=download)
  File "C:\Users\jlopezt\.platformio\.cache\tmp-43287b5n17Z65lk4\virtualenv-16.7.5\virtualenv.py", line 1019, in install_wheel
    _install_wheel_with_search_dir(download, project_names, py_executable, search_dirs)
  File "C:\Users\jlopezt\.platformio\.cache\tmp-43287b5n17Z65lk4\virtualenv-16.7.5\virtualenv.py", line 1046, in _install_wheel_with_search_dir
    config = _pip_config(py_executable, python_path)
  File "C:\Users\jlopezt\.platformio\.cache\tmp-43287b5n17Z65lk4\virtualenv-16.7.5\virtualenv.py", line 1124, in _pip_config
    raise_on_return_code=False,
  File "C:\Users\jlopezt\.platformio\.cache\tmp-43287b5n17Z65lk4\virtualenv-16.7.5\virtualenv.py", line 921, in call_subprocess
    env=env,
  File "D:\Archivos de programa\Python\Python37\Lib\subprocess.py", line 756, in __init__
    restore_signals, start_new_session)
  File "D:\Archivos de programa\Python\Python37\Lib\subprocess.py", line 1155, in _execute_child
    startupinfo)
PermissionError: [WinError 5] Acceso denegado

Please use the *system* python to run this script
Error [WinError 5] Acceso denegado while executing command C:\Users\jlopezt\.platformio\penv\Scripts -m pip config list

	at t.value (c:\Users\jlopezt\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\platformio-node-helpers\dist\index.js:1:31671)
```

ivankravets, 2019-10-12:
It seems like Antivirus or OS blocks PlatformIO. Please install the latest Python 3.7 manually and restart VSCode. See http://docs.platformio.org/en/latest/faq.html#install-python-interpreter

closed 2019-10-12 by ivankravets

jlopezt, 2019-10-12:
Solved!

Cache deleted, system restarted and install alain.

Thx!