platformio/platformio-vscode-ide #1090: Installation Manager

MarkTolra, 2019-10-12:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.1
PIO IDE: v1.9.0
System: Windows_NT, 10.0.17763, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received type undefined
	at t.value (c:\Users\markt\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\platformio-node-helpers\dist\index.js:1:31671)
```

ivankravets, 2019-10-12:
Could you provide installer log from `c:\Users\pc\.platformio\.cache` dir? What is your Python version?
```
python --version
```

ivankravets, 2019-10-18:
Please reopen if you still need help

closed 2019-10-18 by ivankravets

MarkTolra, 2019-10-18:
I managed to solve by completely deleting the .platformio folder and reinstalling.

ivankravets, 2019-10-18:
So, you just deleted `.platformio` folder and restarted VSCode?

MarkTolra, 2019-10-18:
From memory, I uninstalled VSCode, python, and deleted .plaformio basically anything to do with it and started clean. I can't remember what order I did things but clearing .platformio folder got everything working again

ivankravets, 2019-10-18:
> uninstalled python

this is actually the step which helped you. You had broken Python.