platformio/platformio-vscode-ide #1094: Start PIO Home Server

ukrduino, 2019-10-12:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.1
PIO IDE: v1.9.0
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at /Users/sergey/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:17383
	at ChildProcess.h (/Users/sergey/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe._handle.close (net.js:606:12)
```

ivankravets, 2019-10-12:
What is your?
```
python --version
python3 --version
```

ukrduino, 2019-10-12:
Python 2.7.16

I have no Python 3

ivankravets, 2019-10-12:
libcrypto.dll was broken in macOS Catalina. Just re-install official Python 2.7.16 from https://www.python.org/downloads/

Remove `~./platformio/penv` folder, restart VSCode.

Успіхів! 😊

closed 2019-10-12 by ivankravets

ukrduino, 2019-10-12:
Дякую друже, сапорт на вищому ривни!

ivankravets, 2019-10-12:
Завелось?

ukrduino, 2019-10-12:
Да, как раз попробовал.