platformio/platformio-vscode-ide #1095: Update to 1.9.0 may or may not have broken install

pfeerick, 2019-10-12:
Updating to 1.9.0 broke badly, but since I'm also unable to revert to 1.8.3, I'm suspecting it's something else to blame. Manjaro Linux 18.1, which has just received it's latest round of updates, on VSCode 1.39.1

```
console.ts:137 [Extension Host] runCommand /home/pfeerick/.platformio/penv/bin/python3 (5) ["-m", "pip", "install", "-U", "platformio"] {}
console.ts:137 [Extension Host] runCommand platformio (6) ["-f", "-c", "vscode", "home", "--host", "__do_not_start__"] {}
console.ts:137 [Extension Host]  Traceback (most recent call last):
  File "/home/pfeerick/.local/bin/platformio", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/home/pfeerick/.platformio/penv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3191, in <module>
    @_call_aside
  File "/home/pfeerick/.platformio/penv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3175, in _call_aside
    f(*args, **kwargs)
  File "/home/pfeerick/.platformio/penv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3204, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/home/pfeerick/.platformio/penv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 583, in _build_master
    ws.require(__requires__)
  File "/home/pfeerick/.platformio/penv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/home/pfeerick/.platformio/penv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'platformio==4.0.3' distribution was not found and is required by the application
```

ivankravets, 2019-10-12:
> /home/pfeerick/.local/bin/platformio

Do you have multiple PIO Cores in the system? PlatformIO Core should be here `/home/pfeerick/.platformio/penv/bin/platformio`. However, I see it is also installed in  `/home/pfeerick/.local/bin/platformio`.

Could you try to remove multiple versions? http://docs.platformio.org/en/latest/faq.html#multiple-pio-cores-in-a-system

pfeerick, 2019-10-12:
Hm... that is weird... on this Manjaro desktop I only have the stable release running... it's never seen more than one PIO core... and the /home/pfeerick/.local/bin/pio is new... if I delete the ~/.local/bin, ~/.local/lib folders and ~/.platformio folders all three get recreated. Could it be a side-effect of [this recent change in behaviour of pip on Manjaro](https://forum.manjaro.org/t/call-for-testing-making-pip-default-to-user/100682)? 

I also just finished updating my linux (Ubuntu) and Windows 10 installs on the laptop, and they both installed fine.

pfeerick, 2019-10-12:
Damn... it is... it looks like they introduced a bug :-1: 

For anyone else who encounters this issue... you need to edit `/etc/pip.conf` and change `user = true` to `user = false`.

Is the use of `--user` with `pip` expected to break this way with PlatformIO?

ivankravets, 2019-10-12:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/1007

Let's move the discussion to https://github.com/platformio/platformio-vscode-ide/issues/1007

closed 2019-10-12 by ivankravets