platformio/platformio-vscode-ide #1097: Start PIO Home Server

WRvD, 2019-10-12:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.1
PIO IDE: v1.9.0
System: Darwin, 17.7.0, x64

%23 Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at /Volumes/MacintoshSSD2/Users/rob/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:17383
	at ChildProcess.h (/Volumes/MacintoshSSD2/Users/rob/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

geerdkakes, 2019-10-13:
Same issue for me. All worked before. 

Tried the following:
- remove module platformio
- remove folder: `rm -Rf .platformio/penv`
- remove vscode
- reinstall all

Same issue persists.

VSCode: 1.39.1
PIO IDE: v1.9.0
System: Darwin, 18.6.0, x64

%23 Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at /Users/geerd/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:17383
	at ChildProcess.h (/Users/geerd/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

EDIT:
Might be unrelated but when trying to build a build a project I get the following error:

```
CorePackageManager: Installing tool-scons @ ~2.20501.7
Error: Detected unknown package 'tool-scons'
```

ivankravets, 2019-10-13:
Yes, PlatformIO can't download any packages. Is your Internet connection OK? Proxy? What is `python --version`?

Maybe an issue with Python's SSL module.

WRvD, 2019-10-13:
Internet OK
No Proxy
Python version : 2.7.16

ivankravets, 2019-10-13:
Please open PlatformIO IDE Terminal and type:
```
pio home
```

PlatformIO will print an error why it can't download a package.

WRvD, 2019-10-13:
Tried that already.
Result is :

CorePackageManager: Installing contrib-pysite @ ~2.35.190418
Error: Detected unknown package 'contrib-pysite'

ivankravets, 2019-10-13:
Do you see output from this command?
```
python -c "import requests; print(requests.get('https://api.platformio.org/stats').text)"
```

WRvD, 2019-10-13:
No,

Python does not seem to know about a requests module

Rob@imac-fritz-box-1 ~ % python -c "import requests; print(requests.get('https://api.platformio.org/stats').text)"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named requests
Rob@imac-fritz-box-1 ~ %

ivankravets, 2019-10-13:
See updated code here https://github.com/platformio/platformio-vscode-ide/issues/1097#issuecomment-541404238

geerdkakes, 2019-10-13:
Solved my problem by going back to an older version of python (3.7.3)  and platformio (4.0.3) (not sure if the python downgrade was really necessary):
```
brew unlink python
brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/22c80fc362ac8f87c59c446a85a97cb99ff160fb/Formula/python.rb
pip uninstall platformio
pip3 uninstall platformio
pip3 install 'platformio<=4.03'
```

WRvD, 2019-10-14:
> See updated code here #1097 (comment) <https://github.com/platformio/platformio-vscode-ide/issues/1097#issuecomment-541404238>

I don’t understand your advise; #1097 is the issue number I got for my problem.

But….

I solved my problem by following partly another advise.

For OS Mojave removing ~/.platformio/penvin a terminal andrestarting VSCODE solved it

and

For OS Catalina removing ~/.platformio/penv was not enough. I upgraded Python3 to the latest version as well and the problem was solved.

I don’t know whether the upgrade was really necessary; may be just a reinstall might be enough. 

Anyway thanks for the help, I am happy coding again.

closed 2019-10-14 by WRvD