platformio/platformio-vscode-ide #1100: Start PIO Home Server

rmk-8, 2019-10-12:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.1
PIO IDE: v1.9.0
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at /Users/Ralf1/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:17383
	at ChildProcess.h (/Users/Ralf1/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

geerdkakes, 2019-10-12:
Same issue for me. All worked before. 

Tried the following:
- remove module platformio
- remove folder: `rm -Rf .platformio/penv`
- remove vscode
- reinstall all

Same issue persists.

VSCode: 1.39.1
PIO IDE: v1.9.0
System: Darwin, 18.6.0, x64

%23 Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at /Users/geerd/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:17383
	at ChildProcess.h (/Users/geerd/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-10-13:
What is your Python version?
```
python --version
python3 --version
```

rmk-8, 2019-10-13:
I'm really new to this. So i even don't know wether python is installed on my mac. Where can i have a look at this? Knowing python only from my Raspberry PI.

rmk-8, 2019-10-13:
Well, ok, its python 2.7 and python 3.7.3 and clang 11.0.0

ivankravets, 2019-10-14:
It seems you installed Python from Homebrew. Try to uninstall it and install again. Of course, `rm -rf ~/.platformio/penv`. Start VSCode.

closed 2019-10-14 by ivankravets