platformio/platformio-vscode-ide #1101: Start PIO Home Server

Mythkd, 2019-10-12:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.1
PIO IDE: v1.9.0
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at /Users/yuriyvoloshin/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:17383
	at ChildProcess.h (/Users/yuriyvoloshin/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-10-13:
What is your Python version?
```
python --version
python3 --version
```

ivankravets, 2019-10-13:
Could you try the next?

1. Reinstall Python from the source which you used before:
    a) if it's `brew`, then `brew uninstall python`, `brew install python`
    b) if it's Python from Apple Store, just uninstall and install again
    c) if it's official Python from https://python.org, just download the latest Python 3.7 and re-install it.

2. Open system terminal and type `rm -rf ~/.platformio/penv`
3. Restart VScode.

Does it work now?

ivankravets, 2019-10-17:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/1108

See solutions in https://github.com/platformio/platformio-vscode-ide/issues/1108

closed 2019-10-17 by ivankravets