platformio/platformio-vscode-ide #1102: Error: Start PIO Home Server

rkruetli, 2019-10-12:
---- Description of problem
Installing PlatformIO on macOS Catalina under VS Code with Homebrow installed Python 3.7.4 produces the following >> error (1). "Python quit unexpectedly." when PIO Home tries to load >> error (2). PIO Home can not be started.

---- Configuration
VSCode: 1.39.1
PIO IDE: v1.9.0
System: Darwin, 19.0.0, x64

---- Exceptions
```
Error 1: 
	at /Users/xxx/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:17383
	at ChildProcess.h (/Users/rkruetli/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

Error 2:
```
Process:               Python [6104]
Path:                  /usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               3.7.4 (3.7.4)
Code Type:             X86-64 (Native)
Parent Process:        Code Helper [6060]
Responsible:           Electron [6051]
User ID:               501

Date/Time:             2019-10-13 01:29:18.140 +0200
OS Version:            Mac OS X 10.15 (19A583)
Report Version:        12
Anonymous UUID:        487F6ABD-CCA4-4E08-9BAB-1F54885BB866

Time Awake Since Boot: 6200 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
/usr/lib/libcrypto.dylib
abort() called
Invalid dylib load. Clients should not load the unversioned libcrypto dylib as it does not have a stable ABI.
```

ivankravets, 2019-10-13:
Could you try removing Python 3 and installing it again?

ivankravets, 2019-10-13:
Could you try the next?

1. Reinstall Python from the source which you used before:
    a) if it's `brew`, then `brew uninstall python`, `brew install python`
    b) if it's Python from Apple Store, just uninstall and install again
    c) if it's official Python from https://python.org, just download the latest Python 3.7 and re-install it.

2. Open system terminal and type `rm -rf ~/.platformio/penv`
3. Restart VScode.

Does it work now?

rkruetli, 2019-10-13:
Hi Ivan

Sorry for my delay... Ok, seems to be an issue with Homebrew's Python 3.7.4 - with Apples Python 3.7.3  there is **no** error/issue. Have to try it further with the official 3.7.4 version from python.org.  Something strange with the dylip.

Therefore, do not use Homebrews Python 3.7.4 (or do not let it load in VS Code) for the moment.

Thanks!

ivankravets, 2019-10-14:
Hm... Homebrew's Python also should work. Could try to uninstall it and install again? Of course, `rm -rf ~/.platformio/penv`

rmk-8, 2019-10-17:
Hi Ivan,

thanks very much. It worked. Got latest stable python from python.org. It did not work with the original Apple python. But testing your workaround including rm -rf ~/.platformio/pe made it :-)

ivankravets, 2019-10-17:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/1108

See solutions in https://github.com/platformio/platformio-vscode-ide/issues/1108

closed 2019-10-17 by ivankravets