platformio/platformio-vscode-ide #1105: Start PIO Home Server

chrisgish, 2019-10-13:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.1
PIO IDE: v1.9.0
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at /Users/chrisgish/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:17383
	at ChildProcess.h (/Users/chrisgish/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-10-13:
What is your Python version?
```
python --version
python3 --version
```

EnviousD, 2019-10-13:
same problem here

EnviousD, 2019-10-13:
at ChildProcess.emit (events.js:182:13)	at maybeClose (internal/child_process.js:962:16)	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
t.log @ /Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/workbench.desktop.main.js:243
$ pio update

ivankravets, 2019-10-13:
Did you install Python from using Homebrew? Try
```
brew uninstall python
brew install python
```

EnviousD, 2019-10-13:
do I enter these commands into the Console? or the command pallet?

ivankravets, 2019-10-13:
System terminal. Stop, do you use `brew`?

EnviousD, 2019-10-13:
I don't use /brew, should I get myself better aquatinted? I can do some research as I don't have a clue what it is :( building knowledge as I go

ivankravets, 2019-10-13:
Did you install Python manually? Please provide output from
```
python --version
python3 --version
```

EnviousD, 2019-10-13:
Thanks so much for your quick replies, honestly a legend and really appreciate your time!

Python 2.7.16
Python 3.7.3

ivankravets, 2019-10-13:
You don't answer my questions. Did you install Python manually from Python.org? Try downloading https://www.python.org/ftp/python/3.7.4/python-3.7.4-macosx10.9.pkg and reinstall it.

Them, remove `~/.platformio/penv` folder and restart VSCode.

EnviousD, 2019-10-13:
also, I recently updated my OS to Catalina 10.15

EnviousD, 2019-10-13:
I installed python via the extensions marketplace. Should I install it via the website manually, instead?

ivankravets, 2019-10-13:
Could you uninstall it and install again? You can use Marketplace.

EnviousD, 2019-10-13:
I will try now

EnviousD, 2019-10-13:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.40.0-insider
PIO IDE: v1.9.0
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at /Users/alexandersmith/.vscode-insiders/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:17290
	at ChildProcess.h (/Users/alexandersmith/.vscode-insiders/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:6057)
	at ChildProcess.emit (events.js:200:13)
	at maybeClose (internal/child_process.js:1021:16)
	at Socket.<anonymous> (internal/child_process.js:430:11)
	at Socket.emit (events.js:200:13)
	at Pipe.<anonymous> (net.js:586:12)
```

ivankravets, 2019-10-13:
You need to remove this folder `~/.platformio/penv`. Restart VSCode.

EnviousD, 2019-10-13:
I'm just searching for the file now :)

EnviousD, 2019-10-13:
Can't seem to find it, is there a command to delete it via the VSCode terminal?

ivankravets, 2019-10-13:
Run from VSCode terminal
```
rm -rf ~/.platformio/penv
```

Restart VSCode.

EnviousD, 2019-10-13:
it worked!!!!!! You're the BEST!!

ivankravets, 2019-10-13:
So, what you did?

ivankravets, 2019-10-13:
Did you just re-install Python from Apple Store?

EnviousD, 2019-10-13:
Negative. I uninstalled all the python extensions via the marketplace in VSCode. then, I restarted VSCode, went to the "Go" Tab at the top of the screen, "new terminal" entered: rm -rf ~/.platformio/penv

Restarted VSCode again, Re-installed Python first, THEN Python VSCode extensions.

After that It actually successfully started installing PlatformIO IDE properly when I Re-Opened VSCode and went to the Home Button at the bottom left in the Toolbar.

Thanks for much for all the support. Amazing platform, best I've found yet.

ivankravets, 2019-10-13:
Happy coding with PlatformIO!

closed 2019-10-13 by ivankravets