platformio/platformio-vscode-ide #1107: Start PIO Home Server

ursamir, 2019-10-13:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.1
PIO IDE: v1.9.0
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at /Users/sam/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:17383
	at ChildProcess.h (/Users/sam/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-10-13:
Could you try the next?

1. Reinstall Python from the source which you used before:
    a) if it's `brew`, then `brew uninstall python`, `brew install python`
    b) if it's Python from Apple Store, just uninstall and install again
    c) if it's official Python from https://python.org, just download the latest Python 3.7 and re-install it.

2. Open system terminal and type `rm -rf ~/.platformio/penv`
3. Restart VScode.

Does it work now?

ursamir, 2019-10-14:
You are the best man!!!

It worked.

I added a few more steps for safety.
>> I was not sure but my terminal was saying that the python 2.7 is installed. and it seems python 2.7 is provided as a standard library package to support Mac OS backend.
>> Advice [Do not delete this - possible OS failure]
>> But to be safe not any other 2.7 instance of python was installed. I ran few command as follows.
sudo rm -rf /Library/Frameworks/Python.framework/Versions/2.7
sudo rm -rf "/Applications/Python 2.7"
ls -l /usr/local/bin | grep '../Library/Frameworks/Python.framework/Versions/2.7'
cd /usr/local/bin/
ls -l /usr/local/bin | grep '../Library/Frameworks/Python.framework/Versions/2.7' | awk '{print $9}' | tr -d @ | xargs rm

>> Then I installed the python 3.7 from official site.
>> Check what are the python package present.
pkgutil --pkgs | grep org.python.Python

>> It showed me only 3.7 variant of python.
>> then I followed your step of removing platform io from vscode and restarted and waited for reinstallation of platform io.

>> Walah. Its now working.!!!

Thanks man!!!

ursamir, 2019-10-14:
Issue solved. See the above comment.

closed 2019-10-14 by ursamir