platformio/platformio-vscode-ide #1108: Start PIO Home Server

EnviousD, 2019-10-13:
%23 Description of problem
Not loading the PIO Homepage to begin ANY project. Ridiculous, works perfectly on my Microsoft, but this RUBISH 2019 MacBook Pro isn't loading, spent days trying to configure this properly....Rubbisbh apple products....

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.1
PIO IDE: v1.9.0
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at /Users/alexandersmith/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:17383
	at ChildProcess.h (/Users/alexandersmith/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe._handle.close (net.js:606:12)
```

ivankravets, 2019-10-13:
How did you install Python? Did you use Homebrew?

ivankravets, 2019-10-13:
Could you try the next?

1. Reinstall Python from the source which you used before:
    a) if it's `brew`, then `brew uninstall python`, `brew install python`
    b) if it's Python from Apple Store, just uninstall and install again
    c) if it's official Python from https://python.org, just download the latest Python 3.7 and re-install it.

2. Open system terminal and type `rm -rf ~/.platformio/penv`
3. Restart VScode.

Does it work now?

judismith, 2019-10-14:
This is what fixed the issue for me:

On MacOS 10.15, leaving the apple installed python in place, I installed Python 3 from Homebrew. I then executed the `-rf ~/.platformio/penv`. When I restarted VSCode, Python was crashing. The issue was an incompatibility with the crypto libraries. Based on the instructions found at https://stackoverflow.com/questions/58272830/python-crashing-on-macos-10-15-beta-19a582a-with-usr-lib-libcrypto-dylib, I installed openssl from homebrew and then created the necessary symlinks.

PIO homepage now works as expected.

Qube2org, 2019-10-14:
I've been facing same issue on fresh install of VScode on fresh install of Calalina. I've followed the suggestion from ivankravets and run `rm -rf ~/.platformio/penv` and restart VSCode. After PIO was reinstalling itself everything is good. No need to install anything from Homebrew to solve this issue nor do I need to reinstall the python from apple.

edwinm, 2019-11-23:
I had the same problem and removed and reinstalled python as in solution 1.

Then I got error messages about conda.

After running `conda update conda`, PlatformIO did install!

closed 2020-06-03 by ivankravets