platformio/platformio-vscode-ide #1114: Failed to install Platformio IDE in vscode

BowmanChow, 2019-10-14:
## Problem
- I'm using platformio IDE in vscode.

- Before 2019.10.11, the platformio IDE extension worked well.

- But after 2019.10.11, every time I open vscode I get this message.

```
Installing PlatformIO Core...
Please do not close this window and do not open other folders until this process is completed.
Failed to install PlatformIO IDE.
```
- No more information shows.

## What I Have Tried

- Uninstall vscode and reinstall it

- Uninstall platformio IDE and reinstall it

- Downgrade vscode from 1.39 to 1.37

- Downgrade platfotmio IDE from 1.90 to 1.83

- pip --no-cache-dir install -U platformio

- conda install platformio

- pip uninstall platform and then pip install platformio

- Restart the computer

- Uninstall platformio IDE and delete all the folders and files whose names contain 'platformio', and than reinstall platformio

- Run vscode as administrator and install platformio IDE

- I still haven't fixed the problem now.

## What I Have Found

- After I uninstall platformio IDE and tried to reinstall it (of course I failed), I found the folder "C:\Users\Bowman.platformio"'s size is only 0KB, and it contains only one folder ".cache".

## What I Have Installed in My Computer

- Anaconda(Python 3.7)

- JDK

- node.js

- .NET Core

- mingw64

- Visual Studio 2019

- Visual Studio 2017

- Stm32CubeIDE

edalongeville, 2019-10-15:
I have the exact same problem.
Here's what VSCode says:
```
Traceback (most recent call last):
  File "/home/eric/.local/bin/platformio", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/home/eric/.platformio/penv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3191, in <module>
    @_call_aside
  File "/home/eric/.platformio/penv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3175, in _call_aside
    f(*args, **kwargs)
  File "/home/eric/.platformio/penv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 3204, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/home/eric/.platformio/penv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 583, in _build_master
    ws.require(__requires__)
  File "/home/eric/.platformio/penv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/home/eric/.platformio/penv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'platformio==4.0.3' distribution was not found and is required by the application
```

ivankravets, 2019-10-18:
@BowmanChow It can be linked with broken Python installation. Please open VSCode > Help > Developer tools > Console. Do you see any errors?

BowmanChow, 2019-10-18:
@ivankravets Thank you very much. Now I can see errors.  

The error occurs when running commands:
```
conda create --yes --quiet python=2 pip --prefix C:\Users\Bowman\.platformio\penv
```

The errors are as follows

```
Error: 
Conda Virtualenv:  
[WinError 127] The specified procedure could not be found 
[WinError 127] The specified procedure could not be found 
[WinError 127] The specified procedure could not be found 
[WinError 127] The specified procedure could not be found 
[WinError 127] The specified procedure could not be found 
[WinError 127] The specified procedure could not be found 
[WinError 127] The specified procedure could not be found 
[WinError 127] The specified procedure could not be found 
[WinError 127] The specified procedure could not be found   	at o.runCommand 
(c:\Users\Bowman\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\platformio-node-helpers\dist\index.js:1:29676) 	at ChildProcess.h 
(c:\Users\Bowman\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\platformio-node-helpers\dist\index.js:1:6150) 	at ChildProcess.emit (events.js:182:13) 	at ChildProcess.cp.emit 
(c:\Users\Bowman\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\c...
```

ivankravets, 2019-10-18:
@BowmanChow can you fix your conda installation? It's better to remove it and never use again. Please use original and official Python interpreter from https://www.python.org/

So, uninstall this conda and use Python 3.8. Please don't forget to check box with adding Python to PATH.

closed 2019-10-18 by ivankravets

BowmanChow, 2019-10-19:
@ivankravets Thank you very much. I have fixed my conda installation. Your suggestions are really helpful.

jaime385, 2019-12-02:
I do really appreciate the help of both of you guys... In my case, I had the same issue of BowmanChow  but I had to open the command prompt (since I am using windows) and delete manually a carpet just like this:

######################## Examples
# Windows
# C:\Users\UserName\.platformio

When I did so, In addition to installing the official python version (Link: https://www.python.org/downloads/) because I had also installed the Conda interpreter of python and uninstalled it the problem was solved. 

Greetings guys. Thanks a lot.

XiaoboD, 2020-03-08:
@BowmanChow What do you mean by "fix conda installation"? Does it means uninstall it? What should I do if I don't want to uninstall conda?

ivankravets, 2020-03-08:
@XiaoboD do you use the latest conda? Please upgrade it.