platformio/platformio-vscode-ide #1123: Installation Manager

eyer1951, 2019-10-14:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.38.1
PIO IDE: v1.9.0
System: Windows_NT, 10.0.18362, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\Mark\.platformio\.cache\tmp-6800jAiOh8iDHowH\virtualenv-16.7.5\virtualenv.py", line 2632, in <module>
    main()
  File "C:\Users\Mark\.platformio\.cache\tmp-6800jAiOh8iDHowH\virtualenv-16.7.5\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\Users\Mark\.platformio\.cache\tmp-6800jAiOh8iDHowH\virtualenv-16.7.5\virtualenv.py", line 1173, in create_environment
    install_wheel(to_install, py_executable, search_dirs, download=download)
  File "C:\Users\Mark\.platformio\.cache\tmp-6800jAiOh8iDHowH\virtualenv-16.7.5\virtualenv.py", line 1019, in install_wheel
    _install_wheel_with_search_dir(download, project_names, py_executable, search_dirs)
  File "C:\Users\Mark\.platformio\.cache\tmp-6800jAiOh8iDHowH\virtualenv-16.7.5\virtualenv.py", line 1046, in _install_wheel_with_search_dir
    config = _pip_config(py_executable, python_path)
  File "C:\Users\Mark\.platformio\.cache\tmp-6800jAiOh8iDHowH\virtualenv-16.7.5\virtualenv.py", line 1124, in _pip_config
    raise_on_return_code=False,
  File "C:\Users\Mark\.platformio\.cache\tmp-6800jAiOh8iDHowH\virtualenv-16.7.5\virtualenv.py", line 921, in call_subprocess
    env=env,
  File "C:\Program Files\Python36\Lib\subprocess.py", line 709, in __init__
    restore_signals, start_new_session)
  File "C:\Program Files\Python36\Lib\subprocess.py", line 997, in _execute_child
    startupinfo)
PermissionError: [WinError 5] Access is denied

Please use the *system* python to run this script
Error [WinError 5] Access is denied while executing command C:\Users\Mark\.platformio\penv\Scripts -m pip config list

	at t.value (c:\Users\Mark\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\platformio-node-helpers\dist\index.js:1:31671)
```
Problems started after seeing a warning about having two PIO CLI versions installed. Went to the recommended page, uninstalled via pip. Updated PATH variable, etc.  PATH now includes:
`PATH=C:\Users\Mark\.platformio\penv\Scripts;`

ivankravets, 2019-10-17:
Please try to uninstall all Pythons from system and install the latest 3.8. Restart VSCode. Everything should work.

closed 2019-10-17 by ivankravets

eyer1951, 2019-10-20:
I was able to resolve this a different way: delete the .platformio directory, re-install PlatformIO, including all the libraries.