platformio/platformio-vscode-ide #1125: Start PIO Home Server

jessepietila, 2019-10-14:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.38.1
PIO IDE: v1.8.3
System: Darwin, 16.7.0, x64

%23 Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at S.Promise (/Users/jessepietila/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:17372)
	at ChildProcess.h (/Users/jessepietila/.vscode/extensions/platformio.platformio-ide-1.8.3/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

jessepietila, 2019-10-14:
what to do ?

jessepietila, 2019-10-14:
My goal is to install Marlin firmware for Anet A8+ / 3D printer..

ivankravets, 2019-10-18:
Please provide an output from a system Terminal
```
echo $PATH
python --version
pip --version
pip list
```

jessepietila, 2019-10-18:
Thank You Ivan,

I do not know how to do it, please give me some screen shots for advice !

Jesse

ivankravets, 2019-10-18:
Just download the latest Python 3 from official website https://www.python.org/

Install it.

Restart VSCOde.

Does it work now?

jessepietila, 2019-10-18:
[image]

ivankravets, 2019-10-19:
Did you remove `~/.platformio/penv` folder? Please remove it and restart

jessepietila, 2019-10-20:
From where do I remove it ?

ivankravets, 2019-10-20:
Please open macOS Terminal.app and run this command:
```
rm -rf ~/.platformio/penv
```

Restart VSCode.

dewexdewex, 2019-10-21:
Thanks, Ivan. I was having issues with this also. Your instructions fixed it for me.

jessepietila, 2019-10-22:
# Marlin Firmware
# PlatformIO Configuration File

ok this was opened, thank You

I will continue with this later tmrw.. or so

ivankravets, 2019-10-24:
Could you open PlatformIO IDE Terminal, see icon on the bottom toolbar and run:
```
python -c "import requests; print(requests.get('https://api.platformio.org/stats').text)"
```

Do you see the output?

jessepietila, 2019-10-24:
[image]

ivankravets, 2019-10-24:
You need click on TERMINAL icon on the bottom toolbar. It will have valid PATH env.

jessepietila, 2019-10-24:
https://github.com/Microsoft/vscode/issues/44354

I need to add this ?

ivankravets, 2019-10-24:
Do you see PlatformIO icon in the left sidebar?

jessepietila, 2019-10-24:
Yes I do

ivankravets, 2019-10-24:
Left Sidebar > PlatformIO > Misc > PlatformIO Core (CLI):

```
python -c "import requests; print(requests.get('https://api.platformio.org/stats').text)"
```

jessepietila, 2019-10-27:
[image]

jessepietila, 2019-10-27:
[image]

ivankravets, 2019-11-16:
Full line
```
python -c "import requests; print(requests.get('https://api.platformio.org/stats').text)"
```

jessepietila, 2019-11-19:
/Users/jessepietila/Desktop/Näyttökuva 2019-11-19 kello 8.21.20.png

jessepietila, 2019-11-19:
what should happen after this ?

ivankravets, 2019-11-19:
What is your macOS version?

jessepietila, 2019-11-20:
10.13.6

ivankravets, 2019-11-20:
1. Please [upgrade PlatformIO IDE for VSCode to 1.10.0](https://community.platformio.org/t/platformio-ide-for-vscode-1-10/10676) or above
2. Remove `YOUR_USER_HOME_DIR/.platformio/penv` folder
3. Restart VSCode.

If you have a problem with Python interpreter, please uninstall existing [install the latest official Python 3](http://docs.platformio.org/en/latest/faq.html#install-python-interpreter).

closed 2019-11-20 by ivankravets

rubengr, 2020-04-27:
I have the sabe problem @ivankravets .
[image]

ivankravets, 2020-04-27:
It's very strange. Could you try to uninstall this Python 3.8 and install Python 3.7? Also, please remove `HOME_DIR/.platformio/penv` folder and restart VSCode.

rubengr, 2020-04-27:
I will try again, but my previous version is python 3.7, and tried to remove penv several times.

Will try one more..

rubengr, 2020-04-27:
Error: Error: Detected unknown package 'contrib-pysite'

	at c:\Users\...\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:11965
	at ChildProcess.d (c:\Users\...\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\dist\index.js:1:8215)
	at ChildProcess.emit (events.js:203:13)
	at ChildProcess.cp.emit (c:\Users\...\.vscode\extensions\platformio.platformio-ide-1.10.0\node_modules\platformio-node-helpers\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)

ivankravets, 2020-04-27:
Could you try to run this script installer? 
- https://docs.platformio.org/en/latest/core/installation.html#local-download-mac-linux-windows