platformio/platformio-vscode-ide #1127: platformio-ide.* command not found

pfeerick, 2019-10-15:
Reporting for another user. Forum ref: https://community.platformio.org/t/problem-with-platformio-installation-on-vscode/9983/3

Windows 10, VSCode 1.39.1, PIO extension 1.9.0, Python 3.7.3-64bit installed on system

When trying to run (via the Command Palette) the Home, Build or Clean tasks they don't work, and the following errors get generated... Also, it seems the PIO Icons are missing from the sidebar and status bar, and PIO doesn't run from the terminal (although that's probably NOT the PIO terminal)...

```
PlatformIO: Home --> Error: command ‘platformio-ide.home’ not found
PlatformIO: Build --> Error: command ‘platformio-ide.build’ not found
PlatformIO: Clean --> Error: command ‘platformio-ide.clean’ not found
```

Possibly related to #1114, but appears different to #1023.

ivankravets, 2019-10-18:
Thanks! Answered in https://community.platformio.org/t/problem-with-platformio-installation-on-vscode/9983/9?u=ivankravets

closed 2019-10-18 by ivankravets

corymayer, 2019-10-19:
I am having the same issue. Reinstalling Python 3.8 and the extension as recommended above did not work. The PlatformIO toolbar and home works with a new vscode window, but when I create a new project they disappear and the commands no longer work.

foxis, 2019-10-20:
I am having the same issue.
VSCode:
```
Version: 1.39.2 (user setup)
Commit: 6ab598523be7a800d7f3eb4d92d7ab9a66069390
Date: 2019-10-15T15:35:18.241Z
Electron: 4.2.10
Chrome: 69.0.3497.128
Node.js: 10.11.0
V8: 6.9.427.31-electron.0
OS: Windows_NT x64 10.0.18362
```

Before this issue started I was running PlatformIO Tests on one of my projects. The board crashed and was rebooting, so I had to kill the terminal in order to restart the tests.
VSCode froze and I had to restart it.

This happened like 3 or four times and then suddenly there were no icons for PlatformIO.

Tried reinstall the extension - Nothing.
Tried reinstall VSCode - nothing.
Tried uninstalling VSCode and remove all related files, plus `.platformio` and `.vscode`.
Then installing VSCode, and then Platformio IDE. I could see PIO Home.
However right after I add a project directory - Platformio IDE associated icons disappeared, and when I do `Ctrl+Shift+P` I get `platformio-ide.*` errors (depending on chosen command).

Errors in the debug console:
```
[Extension Host] rejected promise not handled within 1 second: TypeError: i.default.isFileSync is not a function
t.log @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
_doInvokeHandler @ rpcProtocol.ts:398
_invokeHandler @ rpcProtocol.ts:383
_receiveRequest @ rpcProtocol.ts:299
_receiveOneMessage @ rpcProtocol.ts:226
_protocol.onMessage.e @ rpcProtocol.ts:101
fire @ event.ts:580
fire @ ipc.net.ts:453
_receiveMessage @ ipc.net.ts:733
_socketDisposables.push._socketReader.onMessage.e @ ipc.net.ts:592
fire @ event.ts:580
acceptChunk @ ipc.net.ts:239
_register._socket.onData.e @ ipc.net.ts:200
t @ ipc.net.ts:28
emit @ events.js:182
addChunk @ _stream_readable.js:283
readableAddChunk @ _stream_readable.js:264
Readable.push @ _stream_readable.js:219
onStreamRead @ internal/stream_base_commons.js:94
console.ts:137 [Extension Host] stack trace: TypeError: i.default.isFileSync is not a function	at Object.t.isPIOProject (c:\Users\Foxis\.vscode\extensions\platformio.platformio-ide-1.9.1\node_modules\platformio-node-helpers\dist\index.js:1:4689)	at map.filter.e (c:\Users\Foxis\.vscode\extensions\platformio.platformio-ide-1.9.1\dist\extension.js:1:13675)	at Array.filter (<anonymous>)	at c (c:\Users\Foxis\.vscode\extensions\platformio.platformio-ide-1.9.1\dist\extension.js:1:13658)	at Object.t.getActivePIOProjectDir (c:\Users\Foxis\.vscode\extensions\platformio.platformio-ide-1.9.1\dist\extension.js:1:12547)	at Object.activate (c:\Users\Foxis\.vscode\extensions\platformio.platformio-ide-1.9.1\dist\extension.js:1:2810)	at t.activate (c:\Users\Foxis\.vscode\extensions\platformio.platformio-ide-1.9.1\dist\extension.js:1:1684)	at Function._callActivateOptional (c:\Users\Foxis\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:667:613)	at Function._callActivate (c:\Users\Foxis\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:667:264)	at Promise.all.then.n (c:\Users\Foxis\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:666:404)
t.log @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
_doInvokeHandler @ rpcProtocol.ts:398
_invokeHandler @ rpcProtocol.ts:383
_receiveRequest @ rpcProtocol.ts:299
_receiveOneMessage @ rpcProtocol.ts:226
_protocol.onMessage.e @ rpcProtocol.ts:101
fire @ event.ts:580
fire @ ipc.net.ts:453
_receiveMessage @ ipc.net.ts:733
_socketDisposables.push._socketReader.onMessage.e @ ipc.net.ts:592
fire @ event.ts:580
acceptChunk @ ipc.net.ts:239
_register._socket.onData.e @ ipc.net.ts:200
t @ ipc.net.ts:28
emit @ events.js:182
addChunk @ _stream_readable.js:283
readableAddChunk @ _stream_readable.js:264
Readable.push @ _stream_readable.js:219
onStreamRead @ internal/stream_base_commons.js:94
mainThreadExtensionService.ts:65 [[object Object]]i.default.isFileSync is not a function
$onExtensionRuntimeError @ mainThreadExtensionService.ts:65
[... 44 lines cut ...]
addChunk @ _stream_readable.js:283
readableAddChunk @ _stream_readable.js:264
Readable.push @ _stream_readable.js:219
onStreamRead @ internal/stream_base_commons.js:94
notificationsAlerts.ts:38 Error: command 'platformio-ide.test' not found
    at Proxy._tryExecuteCommand (commandService.ts:80)
    at r.then.i (commandService.ts:73)
    at process._tickCallback (internal/process/next_tick.js:68)
onDidNotificationChange @ notificationsAlerts.ts:38
_register.model.onDidNotificationChange.e @ notificationsAlerts.ts:26
fire @ event.ts:580
addNotification @ notifications.ts:156
error @ notificationService.ts:56
onError @ commandsHandler.ts:332
setTimeout @ commandsHandler.ts:319
_tickCallback @ internal/process/next_tick.js:68
async function (async)
setTimeout @ commandsHandler.ts:310
setTimeout (async)
runAction @ commandsHandler.ts:303
run @ commandsHandler.ts:287
run @ quickOpenModel.ts:575
elementSelected @ quickOpenWidget.ts:561
_register.u.addDisposableListener.e @ quickOpenWidget.ts:235
```