platformio/platformio-vscode-ide #1129: UnicodeEncodeError: 'gbk' codec

richard-ubi, 2019-10-15:
hi friends！
I suddenly have problems with compilation failure during use, which makes it impossible to compile.

`
UnicodeEncodeError: 'gbk' codec can't encode character '\xbf' in position 40285: illegal multibyte sequence:
  File "C:\Users\Administrator\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
......
`

ivankravets, 2019-10-18:
1. Please uninstall all Python interpreters from system.
2. Install the latest Python 3.8. See http://docs.platformio.org/en/latest/faq.html#install-python-interpreter
3. Check that antivirus tools do not block PlatformIO.
4. Restart VSCode.

Do you still have this issue?

ivankravets, 2019-10-18:
Duplicate of https://github.com/platformio/platformio-core/issues/3143

closed 2019-10-18 by ivankravets

richard-ubi, 2019-10-19:
Thank you very much, I re-worked the platform by reinstalling VScode and configuring python development environment.

ivankravets, 2019-10-19:
Did you reinstall Python?

richard-ubi, 2019-10-21:
Yes, it may be that Python has updated the version itself before it is unsuccessful.