platformio/platformio-vscode-ide #1139: MacOS Catalina error: Start PIO Home Server

joechin, 2019-10-15:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.38.1
PIO IDE: v1.9.0
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at S.Promise (/Users/joseph/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:17383)
	at ChildProcess.h (/Users/joseph/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe.Socket._destroy._handle.close (net.js:606:12)
```

joechin, 2019-10-15:
So I found the solution based. MacOS has deprecated OpenSSL support with their own.

`$ brew install ssl`
`$ cd /usr/local/lib`
`$ ln -s ../Cellar/openssl/1.0.2t/lib/libssl.1.0.0.dylib libssl.dylib`
`$ ln -s ../Cellar/openssl/1.0.2t/lib/libcrypto.1.0.0.dylib libcrypto.dylib`
`$ cd`
`$ cd .platformio/`
`$ rm -rf penv`

Restart VSCode and open the PlatformIO workspace again.

Mogera, 2019-10-15:
Thanks for sharing your solution @joechin , this resolved the issue on my end.
I had to run 'brew install openssl' first to get the right components installed.

I am by no means an expert but the current extension implementation feels quite brittle.
Looking at the amount of incoming new issues I expect quite a few people are impacted.

ivankravets, 2019-10-17:
Thanks for the solution. I updated main issue https://github.com/platformio/platformio-vscode-ide/issues/1108

closed 2019-10-17 by ivankravets