platformio/platformio-vscode-ide #1142: Start PIO Home Server

vlapa1973, 2019-10-15:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.38.1
PIO IDE: v1.9.0
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at S.Promise (/Users/vlapa./.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:17383)
	at ChildProcess.h (/Users/vlapa./.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe.Socket._destroy._handle.close (net.js:606:12)
```

haukero, 2019-10-15:
Try installing an updated and fresh Python version, this solved it for me.
https://www.python.org/downloads/release/python-2716/

luc-debug, 2019-10-16:
Thank you so much that solved the issue for me.
The problem was that I installed Python 3.8.0. But with your suggested version it works fine.

ivankravets, 2019-10-17:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/1108

See solutions in https://github.com/platformio/platformio-vscode-ide/issues/1108

ivankravets, 2019-10-17:
Python 3.8.0 works too. We have just deployed binaries for PIO Home.

closed 2019-10-17 by ivankravets