platformio/platformio-vscode-ide #1152: Start PIO Home Server

skyflashde, 2019-10-16:
%23 Description of problem
Leave a comment...

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.0
System: Windows_NT, 10.0.17763, x64

%23 Exception
```
Error: Error: Could not find a version that satisfies the requirement '~2.38.190418' for your system 'windows_x86'

	at c:\Users\webma.DESKTOP-U0THE4M\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\platformio-node-helpers\dist\index.js:1:17383
	at ChildProcess.h (c:\Users\webma.DESKTOP-U0THE4M\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\platformio-node-helpers\dist\index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (c:\Users\webma.DESKTOP-U0THE4M\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

mrdoublejr, 2019-10-16:
I have the same EXACT problem

Error: Error: Could not find a version that satisfies the requirement '~2.38.190418' for your system 'windows_x86'

	at c:\Users\mrdou\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\platformio-node-helpers\dist\index.js:1:17383
	at ChildProcess.h (c:\Users\mrdou\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\platformio-node-helpers\dist\index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (c:\Users\mrdou\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)

ivankravets, 2019-10-17:
Please restart VSCode. It should work now. A new package for Python 3.8 has been deployed.

closed 2019-10-17 by ivankravets

mrdoublejr, 2019-10-18:
Beautiful! thanks a million