platformio/platformio-vscode-ide #1154: "UnicodeDecodeError: 'gbk' codec"in build

Coloryr, 2019-10-16:
When i start build,it will show this in the console.

`Compiling .pio\build\genericSTM32F103VE\src\main.o`
`Exception in thread Thread-3:`
`Traceback (most recent call last):`
`  File "C:\Users\Color_yr\.platformio\python37\lib\threading.py", line 926, in _bootstrap_inner`
`    self.run()`
`  File "C:\Users\Color_yr\.platformio\python37\lib\threading.py", line 870, in run`
`    self._target(*self._args, **self._kwargs)`
`  File "C:\Users\Color_yr\.platformio\penv\lib\site-packages\platformio\proc.py", line 42, in run`
`    self.do_reading()`
`  File "C:\Users\Color_yr\.platformio\penv\lib\site-packages\platformio\proc.py", line 66, in do_reading`
`    for byte in iter(lambda: self._pipe_reader.read(1), ""):`
`  File "C:\Users\Color_yr\.platformio\penv\lib\site-packages\platformio\proc.py", line 66, in <lambda>`
`    for byte in iter(lambda: self._pipe_reader.read(1), ""):`
`UnicodeDecodeError: 'gbk' codec can't decode byte 0xbd in position 210: illegal multibyte sequence`

my code have some Chinese character.

ivankravets, 2019-10-18:
1. Please uninstall all Python interpreters from system.
2. Install the latest Python 3.8. See http://docs.platformio.org/en/latest/faq.html#install-python-interpreter
3. Check that antivirus tools do not block PlatformIO.
4. Restart VSCode.

Do you still have this issue? Please check that you save file in UTF-8 encoding.

ivankravets, 2019-10-18:
Also, could you provide a simple project to reproduce this issue?