platformio/platformio-vscode-ide #1158: Start PIO Home Server

ehagerty, 2019-10-16:
This is with standard python, but with pipenv installed...

%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.40.0-insider
PIO IDE: v1.9.0
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: 
	at /Users/ehagerty/.vscode-insiders/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:17290
	at ChildProcess.h (/Users/ehagerty/.vscode-insiders/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:6057)
	at ChildProcess.emit (events.js:200:13)
	at ChildProcess.EventEmitter.emit (domain.js:471:20)
	at maybeClose (internal/child_process.js:1021:16)
	at Socket.<anonymous> (internal/child_process.js:430:11)
	at Socket.emit (events.js:200:13)
	at Socket.EventEmitter.emit (domain.js:471:20)
	at Pipe.<anonymous> (net.js:586:12)
```

dylanmc, 2019-10-16:
The comments on this issue solve it for Catalina (which deprecates a crypto library): https://github.com/platformio/platformio-vscode-ide/issues/1139

ivankravets, 2019-10-17:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/1108

See solutions in https://github.com/platformio/platformio-vscode-ide/issues/1108