platformio/platformio-vscode-ide #1162: Installation Manager

baget, 2019-10-16:
%23 Description of problem
I'm windows store python and PIO does not work, fine with it..

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.40.0-insider
PIO IDE: v1.9.0
System: Windows_NT, 10.0.18362, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\User\.platformio\.cache\tmp-42096iMBTlnKYWhrp\virtualenv-16.7.5\virtualenv.py", line 2632, in <module>
    main()
  File "C:\Users\User\.platformio\.cache\tmp-42096iMBTlnKYWhrp\virtualenv-16.7.5\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\Users\User\.platformio\.cache\tmp-42096iMBTlnKYWhrp\virtualenv-16.7.5\virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\User\.platformio\.cache\tmp-42096iMBTlnKYWhrp\virtualenv-16.7.5\virtualenv.py", line 1526, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.1264.0_x64__qbz5n2kfra8p0\lib\shutil.py", line 120, in copyfile
    with open(src, 'rb') as fsrc:
OSError: [Errno 22] Invalid argument: 'C:\\Users\\User\\AppData\\Local\\Microsoft\\WindowsApps\\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\\python.exe'

Using base prefix 'C:\\Program Files\\WindowsApps\\PythonSoftwareFoundation.Python.3.7_3.7.1264.0_x64__qbz5n2kfra8p0'
New python executable in C:\Users\User\.platformio\penv\Scripts\python.exe

	at t.value (c:\Users\User\.vscode-insiders\extensions\platformio.platformio-ide-1.9.0\node_modules\platformio-node-helpers\dist\index.js:1:31578)
```

ivankravets, 2019-10-18:
1. Please uninstall all Python interpreters from system.
2. Install the latest Python 3.8. See http://docs.platformio.org/en/latest/faq.html#install-python-interpreter
3. Check that antivirus tools do not block PlatformIO.
4. Restart VSCode.

baget, 2019-10-18:
> 1. Please uninstall all Python interpreters from system.
> 2. Install the latest Python 3.8. See http://docs.platformio.org/en/latest/faq.html#install-python-interpreter
> 3. Check that antivirus tools do not block PlatformIO.
> 4. Restart VSCode.

Uninstall all pythons from my system? this is not a solution.
I want to use the python from Windows store, because when I can update quickly
and I only Windows Defender (default windows Antivirus)

ivankravets, 2019-10-18:
Sorry, I’m not Windows user. I’ll try in virtual machine to try Python from Microsoft store.

ivankravets, 2019-10-18:
@baget I can't reproduce this issue, awful windows os does not allow me to install this Python 3.8 from Windows Store. 10 years and Microsoft can't copy basic features from macOS, such as Apple Store -> Windows Store.

So, I'll need your help on this issue. Could you open `CMD.EXE` and type the next?
```
python -m venv C:\testvenv
```

Does it work?

baget, 2019-10-18:
> ```
> python -m venv C:\testvenv
> ```

yes it does:
```

D:\Work>python -m venv D:\work\testvenv                                                                                                                                                                                                                                                                                         
D:\Work>                                                                                                                                                        
D:\Work>dir d:\Work\testvenv                                                                                                                                     
Volume in drive D is Disk2                                                                                                                                      
Directory of d:\Work\testvenv\                                                                                                                                                                                                                                                                                     10/18/2019  10:01 PM    <DIR>          .                                                                                                                        
10/18/2019  10:01 PM    <DIR>          ..                                                                                                                       
10/18/2019  10:01 PM    <DIR>          Include                                                                                                                  
10/18/2019  10:01 PM    <DIR>          Lib                                                                                                                      
10/18/2019  10:01 PM               163 pyvenv.cfg                                                                                                               
10/18/2019  10:01 PM    <DIR>          Scripts                                                                                                                                 
1 File(s)            163 bytes                                                                                                                                  
5 Dir(s)  164,880,003,072 bytes free                                                                                                                                                                                                                                                                             
D:\Work>                                                                                                                                                        

```

ivankravets, 2019-10-18:
This a bug on our side. The Store's Python is installed in a special folder and system environment `PATH` is extended with patterns `%***%` instead of a full path. We will fix it soon.

ivankravets, 2019-10-18:
Could you re-test this build 1.9.1-beta.2? 

1. Download https://www.dropbox.com/s/9p77qe77qrnpzu5/platformio-ide-1.9.1-beta.2.vsix?dl=1
2. VSCode > View > Command Palette > Install from VSIX...
3. Select the downloaded VSIX file.
4. Restart VSCode.

Does it work now?

pekkalehtikoski, 2019-10-19:
Somewhat similar issue on deepin linux 15.11: switching from Python 3.7.4 to Python 3.8 seemed to solve it, VSIX package did not. This is unreliable information, but supports Ivan's "switch to 3.8" proposal.

ivankravets, 2019-10-19:
1.9.1 is out! Please update. 
- https://github.com/platformio/platformio-vscode-ide/releases/tag/v1.9.1