platformio/platformio-vscode-ide #1191: Installation Manager

fblondeau, 2019-10-17:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.0
System: Darwin, 19.0.0, x64

%23 Exception
```
Error: PIP Core:   Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', ConnectionResetError(54, 'Connection reset by peer'))': /simple/platformio/
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', ConnectionResetError(54, 'Connection reset by peer'))': /simple/platformio/
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', ConnectionResetError(54, 'Connection reset by peer'))': /simple/platformio/
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', ConnectionResetError(54, 'Connection reset by peer'))': /simple/platformio/
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'ProtocolError('Connection aborted.', ConnectionResetError(54, 'Connection reset by peer'))': /simple/platformio/
  Could not find a version that satisfies the requirement platformio (from versions: )
No matching distribution found for platformio

	at o.runCommand (/Users/francisblondeau/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:32594)
	at ChildProcess.h (/Users/francisblondeau/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-10-18:
See https://github.com/platformio/platformio-vscode-ide/issues/1108

Please re-install your custom Python.

closed 2019-10-18 by ivankravets

fblondeau, 2019-10-18:
Its not custom. the install is conflicting with homebrew python install. There's should be a way to avoid that. I haven't try to fix it.  if platoformIO or Homebrew get updated one screw will the other. 

In the mean time I reinstall platformio and python to solve the issue until next update.

ivankravets, 2019-10-18:
@fblondeau this is an issue with homebrew's python on catlina.

fblondeau, 2019-10-19:
Il look into it. You can keep this close.