platformio/platformio-vscode-ide #1193: "UnicodeDecodeError: 'gbk' codec"in build

Coloryr, 2019-10-18:
When i start build,it will show this in the console.

`Compiling .pio\build\genericSTM32F103VE\src\main.o`
`Exception in thread Thread-3:`
`Traceback (most recent call last):`
`  File "C:\Users\Color_yr\.platformio\python37\lib\threading.py", line 926, in _bootstrap_inner`
`    self.run()`
`  File "C:\Users\Color_yr\.platformio\python37\lib\threading.py", line 870, in run`
`    self._target(*self._args, **self._kwargs)`
`  File "C:\Users\Color_yr\.platformio\penv\lib\site-packages\platformio\proc.py", line 42, in run`
`    self.do_reading()`
`  File "C:\Users\Color_yr\.platformio\penv\lib\site-packages\platformio\proc.py", line 66, in do_reading`
`    for byte in iter(lambda: self._pipe_reader.read(1), ""):`
`  File "C:\Users\Color_yr\.platformio\penv\lib\site-packages\platformio\proc.py", line 66, in <lambda>`
`    for byte in iter(lambda: self._pipe_reader.read(1), ""):`
`UnicodeDecodeError: 'gbk' codec can't decode byte 0xbd in position 210: illegal multibyte sequence`

my code have some Chinese character.

Coloryr, 2019-10-19:
ok, i try it,now it can build,i use the stlink can upload,but can't debug

ivankravets, 2019-11-16:
> if use "python37",it can work well in debug.
Please file in https://github.com/platformio/platform-ststm32

closed 2019-11-16 by ivankravets