platformio/platformio-vscode-ide #1194: Start PIO Home Server

BLimaM4I, 2019-10-18:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.1
PIO IDE: v1.9.0
System: Windows_NT, 10.0.18362, x64

%23 Exception
```
Error: [WinError 5] Acesso negado: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\cryptography\\hazmat\\bindings\\_constant_time.cp37-win_amd64.pyd' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\cryptography\hazmat\bindings\_constant_time.cp37-win_amd64.pyd`
[WinError 5] Acesso negado: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\cryptography\\hazmat\\bindings\\_openssl.cp37-win_amd64.pyd' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\cryptography\hazmat\bindings\_openssl.cp37-win_amd64.pyd`
[WinError 145] O diret�rio n�o est� vazio: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\cryptography\\hazmat\\bindings' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\cryptography\hazmat\bindings`
[WinError 145] O diret�rio n�o est� vazio: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\cryptography\\hazmat' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\cryptography\hazmat`
[WinError 145] O diret�rio n�o est� vazio: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\cryptography' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\cryptography`
[WinError 5] Acesso negado: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\pywin32_system32\\pywintypes37.dll' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\pywin32_system32\pywintypes37.dll`
[WinError 145] O diret�rio n�o est� vazio: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\pywin32_system32' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\pywin32_system32`
[WinError 5] Acesso negado: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\win32\\win32api.pyd' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\win32\win32api.pyd`
[WinError 5] Acesso negado: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\win32\\win32event.pyd' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\win32\win32event.pyd`
[WinError 5] Acesso negado: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\win32\\win32file.pyd' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\win32\win32file.pyd`
[WinError 5] Acesso negado: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\win32\\win32gui.pyd' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\win32\win32gui.pyd`
[WinError 5] Acesso negado: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\win32\\win32pipe.pyd' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\win32\win32pipe.pyd`
[WinError 5] Acesso negado: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\win32\\win32process.pyd' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\win32\win32process.pyd`
[WinError 5] Acesso negado: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\win32\\_win32sysloader.pyd' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\win32\_win32sysloader.pyd`
[WinError 145] O diret�rio n�o est� vazio: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\win32' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\win32`
[WinError 5] Acesso negado: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\zope\\interface\\_zope_interface_coptimizations.cp37-win_amd64.pyd' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\zope\interface\_zope_interface_coptimizations.cp37-win_amd64.pyd`
[WinError 145] O diret�rio n�o est� vazio: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\zope\\interface' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\zope\interface`
[WinError 145] O diret�rio n�o est� vazio: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\zope' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\zope`
[WinError 5] Acesso negado: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite\\_cffi_backend.cp37-win_amd64.pyd' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite\_cffi_backend.cp37-win_amd64.pyd`
[WinError 145] O diret�rio n�o est� vazio: 'C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite' 
Please manually remove the file `C:\Users\bruno\.platformio\packages\contrib-pysite`
Error: Could not install 'contrib-pysite' with version requirements '~2.37.190418' for your system 'windows_amd64'.

Please try this solution -> http://bit.ly/faq-package-manager

	at c:\Users\bruno\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\platformio-node-helpers\dist\index.js:1:17383
	at ChildProcess.h (c:\Users\bruno\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\platformio-node-helpers\dist\index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (c:\Users\bruno\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

neilh10, 2019-10-18:
I had this happen as well. Since its all well stated, I'm not sure there is any value in adding more info.

I really appreciate PIO and the tools cc management - mostly it just works, but last day or so having a fair number of problems with initializing new libs from github. Can't really track down why, but says its lacking privilege. I try running as admin, and that doesn't solve the problem. I just keep pressing the build icon, and then it seems to get a little further and then eventually does build and I can test my code.
I'm using Win 10 pro, with VSC latest (version 1.39.1 now updateing to 1.29.2)and platformio (Home 2.3.3 Core 4.0.3) and V
I have just switched to default anti virus Win 10 Defender from Bitdefender as Bitdefender was not allowing any real-cpu to be used, and bitdefender support had no viable suggestions.

ivankravets, 2019-10-19:
Yes, antivirus tools sometimes block PlatformIO installer because we call Python from VSCode's NodeJS as a subprocess. Antivirus thinks that we do something bad :(

1. Try to remove `C:\\Users\\bruno\\.platformio\\packages\\contrib-pysite` folder
2. Try to disable antivirus tool for a while
3. Restart VSCode.

Does it work now?

BLimaM4I, 2019-10-21:
Hi,
Removing the folder woked for me.
Thanks,
Bruno

closed 2019-10-21 by ivankravets