platformio/platformio-vscode-ide #1199: Installation Manager

wangyixu12, 2019-10-19:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.0
System: Windows_NT, 10.0.18362, x64

%23 Exception
```
Error: PIP Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while. 
   Could not find a version that satisfies the requirement click<8,>=5 (from platformio) (from versions: )
No matching distribution found for click<8,>=5 (from platformio)

	at o.runCommand (c:\Users\wangy\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\platformio-node-helpers\dist\index.js:1:32594)
	at ChildProcess.h (c:\Users\wangy\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\platformio-node-helpers\dist\index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (c:\Users\wangy\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-10-19:
Do you have antivirus tools in a system?

wangyixu12, 2019-11-06:
I don't have any antivirus in my computer.

ivankravets, 2019-11-16:
Did you fix this issue?

closed 2019-11-21 by ivankravets