platformio/platformio-vscode-ide #1200: Installation Manager

mrunmaydarpan, 2019-10-19:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.0
System: Windows_NT, 10.0.17763, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\mrunm\.platformio\.cache\tmp-101126qVeVKaEiu5D\virtualenv-16.7.5\virtualenv.py", line 2632, in <module>
    main()
  File "C:\Users\mrunm\.platformio\.cache\tmp-101126qVeVKaEiu5D\virtualenv-16.7.5\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\Users\mrunm\.platformio\.cache\tmp-101126qVeVKaEiu5D\virtualenv-16.7.5\virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\mrunm\.platformio\.cache\tmp-101126qVeVKaEiu5D\virtualenv-16.7.5\virtualenv.py", line 1526, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Users\mrunm\AppData\Local\Programs\Python\Python37-32\lib\shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\mrunm\\.platformio\\penv\\Scripts\\python.exe'

Using base prefix 'C:\\Users\\mrunm\\AppData\\Local\\Programs\\Python\\Python37-32'
New python executable in C:\Users\mrunm\.platformio\penv\Scripts\python.exe

	at t.value (c:\Users\mrunm\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\platformio-node-helpers\dist\index.js:1:31671)
```

pekkalehtikoski, 2019-10-19:
Same here. I did not work on this for a few weeks, and now (Saturday) when I opened vscode to return to esp32 software I got this.

pekkalehtikoski, 2019-10-19:
Switching from Python 3.7.4 to 3.8 seemed to solve this problem for me. I am unsure if it is the same problem, error messages look similar and the problem started about same time, but I run Deepin GNU/Linux 15.11.

ivankravets, 2019-10-19:
Antivirus tools can block PlatformIO. try to disable them for a while. Also, if you have broken Python 3 - just re-install it.

closed 2019-10-19 by ivankravets

JZ-SmartThings, 2019-10-23:
On Windows, the key was to kill all python.exe processes... a reboot would achieve the same thing. After that try updating PIO. My issue was because VSCode and PIO tried to update at the same time so had orphan python.exe processes left over from the failed attempt.