platformio/platformio-vscode-ide #1201: Start PIO Home Server - not loading, not build project

Psh777, 2019-10-19:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.0
System: Darwin, 17.7.0, x64

%23 Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

    at /Users/psh/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/home.js:204:28
    at ChildProcess.h (/Users/psh/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/misc.js:188:5)
    at ChildProcess.emit (events.js:182:13)
    at maybeClose (internal/child_process.js:962:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

CorePackageManager: Installing tool-scons @ ~3.30101.0
Error: Detected unknown package 'tool-scons'

Psh777, 2019-10-19:
python ver 3.7.4_1

Psh777, 2019-10-19:
[image]

pekkalehtikoski, 2019-10-19:
Switching to Python 3.8 seems to have solved recent somewhat similar issues for several people, including me. Is this easy for you to try?

Psh777, 2019-10-19:
> Switching to Python 3.8 seems to have solved recent somewhat similar issues for several people, including me. Is this easy for you to try?

[image]

Thanks! It work.

But PIO HOME not loading.

[image]

iocafe, 2019-10-19:
Hi, I do not have Darwin, so this may not be applicable. I had similar issue on deepin Linux and Python update solved that. I read that several people have lately had issues with similar symptoms with <home>.platformio/penv or <home>.platformio/packages directory having some old content. I tried on Windows computer and it had also similar issues. Below my notes what I did to resolve. Hope this helps:
----------------
Uninstall PlatformIO from Visual Studio Code

Uninstall all Python versions from add/remove programs, just to make sure that new one will be used.
Install Python 3.8. I used 64 bit version (Big download button may give 32 bit version even for 64 bit machine), installed it all for all users and let it set set environment variables.

Reboot the computer

On Windows open command prompt and type "python --version", or on linux terminal and type "python3 --version". It needs to be 3.8.0.

Use file explorer to move c:\Users\<yourname>\.platformio or ~/.platformio. The subfolders penv and packages may be messed up. Right click and remove read only, and then it should be possible to delete the penv folder. I am unsure if packages folder needs to be deleted, but may be good idea if you get "The directory is not empty" errs.

Start Visual Stdio Code and install PlatformIO. At this point it installed cleanly for me.
Start "PlatformIO: New terminal" from [View] [Command Palette]. type "pio update"

Seems that there are two known reasons for problems somewhat similar symptoms
- https://github.com/platformio/platformio-core/issues/2341 - something wrong with penv folder
- Other than Python 3.8 version is used, either 3.8 is not installed or also earlier versions are installed.

On Windows I just install one version and remove others. On linux I use "update-alternatives" to select which version python3 starts.

pekkalehtikoski, 2019-10-19:
Ivan seems to be furiously working on this issue. PlatformIO IDE version 1.9.1 was released just half an hour ago, and seems to focus on these issues. Saturday night, what better a young man should do at this hour?

iocafe, 2019-10-19:
Tries PlatformIO IDE 1.9.1 on Linux and Windows. No quick success in either, PlatformIO home doesn't load. Reverted both back to 1.9.0 to continue coding. I see the same screen with "Loading..." (your screen print) after switch to 1.9.0. It stays on for a while, I guess it is working on something, and then all is good.

ivankravets, 2019-10-21:
@Psh777  did you resolve this issue?

Psh777, 2019-10-21:
python --version
Python 2.7.16

python3 --version
Python 3.8.0a2

remove folder ~/.platformio

pio version 1.9.2

PIO HOME also does not load :(

ivankravets, 2019-10-21:
Could you click on PlatformIO IDE terminal icon (bottom bar) and type
```
pio home
```

Do you see any errors?

pekkalehtikoski, 2019-10-21:
Gi Psh777. If you have a change 1.9.2 might well be worth trying. I did not test it on Darwin but it installed cleanly on Linux Deepin 15 and Mint 18. Almost cleanly on Windows, with minor "permission denied #1213" issue, solvable by Windows restart.

ivankravets, 2019-10-23:
Please try this solution https://github.com/platformio/platformio-vscode-ide/issues/1259#issuecomment-545377055

closed 2019-10-23 by ivankravets