platformio/platformio-vscode-ide #1205: Start PIO Home Server

empakoso, 2019-10-19:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.0
System: Windows_NT, 10.0.18362, x64

%23 Exception
```
Error: [WinError 5] Access is denied: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\cryptography\\hazmat\\bindings\\_constant_time.cp37-win_amd64.pyd' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\cryptography\hazmat\bindings\_constant_time.cp37-win_amd64.pyd`
[WinError 5] Access is denied: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\cryptography\\hazmat\\bindings\\_openssl.cp37-win_amd64.pyd' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\cryptography\hazmat\bindings\_openssl.cp37-win_amd64.pyd`
[WinError 145] The directory is not empty: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\cryptography\\hazmat\\bindings' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\cryptography\hazmat\bindings`
[WinError 145] The directory is not empty: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\cryptography\\hazmat' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\cryptography\hazmat`
[WinError 145] The directory is not empty: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\cryptography' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\cryptography`
[WinError 5] Access is denied: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\pywin32_system32\\pywintypes37.dll' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\pywin32_system32\pywintypes37.dll`
[WinError 145] The directory is not empty: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\pywin32_system32' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\pywin32_system32`
[WinError 5] Access is denied: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\win32\\win32api.pyd' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\win32\win32api.pyd`
[WinError 5] Access is denied: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\win32\\win32event.pyd' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\win32\win32event.pyd`
[WinError 5] Access is denied: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\win32\\win32file.pyd' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\win32\win32file.pyd`
[WinError 5] Access is denied: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\win32\\win32gui.pyd' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\win32\win32gui.pyd`
[WinError 5] Access is denied: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\win32\\win32pipe.pyd' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\win32\win32pipe.pyd`
[WinError 5] Access is denied: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\win32\\win32process.pyd' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\win32\win32process.pyd`
[WinError 5] Access is denied: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\win32\\_win32sysloader.pyd' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\win32\_win32sysloader.pyd`
[WinError 145] The directory is not empty: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\win32' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\win32`
[WinError 5] Access is denied: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\zope\\interface\\_zope_interface_coptimizations.cp37-win_amd64.pyd' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\zope\interface\_zope_interface_coptimizations.cp37-win_amd64.pyd`
[WinError 145] The directory is not empty: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\zope\\interface' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\zope\interface`
[WinError 145] The directory is not empty: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\zope' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\zope`
[WinError 5] Access is denied: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite\\_cffi_backend.cp37-win_amd64.pyd' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite\_cffi_backend.cp37-win_amd64.pyd`
[WinError 145] The directory is not empty: 'C:\\Users\\empak\\.platformio\\packages\\contrib-pysite' 
Please manually remove the file `C:\Users\empak\.platformio\packages\contrib-pysite`
Error: Could not install 'contrib-pysite' with version requirements '~2.37.190418' for your system 'windows_amd64'.

Please try this solution -> http://bit.ly/faq-package-manager

	at c:\Users\empak\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\platformio-node-helpers\dist\index.js:1:17383
	at ChildProcess.h (c:\Users\empak\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\platformio-node-helpers\dist\index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at ChildProcess.cp.emit (c:\Users\empak\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

ivankravets, 2019-10-19:
1. Please open PlatformIO IDE Terminal and type `pio update`
2. Restart VSCode.

Did it help?

empakoso, 2019-10-20:
Works good! Thank you. Really new to all this (marlin FW).  Rockin and Rollin - where has PlatformIO been all my life!

closed 2019-10-20 by empakoso