platformio/platformio-vscode-ide #1206: Start PIO Home Server

Death-art, 2019-10-19:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.38.1
PIO IDE: v1.9.0
System: Darwin, 16.7.0, x64

%23 Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at S.Promise (/Users/jk/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:17383)
	at ChildProcess.h (/Users/jk/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)
```

pekkalehtikoski, 2019-10-19:
Switching from Python 3.7.4 to Python 3.8 seems to have solved recent somewhat similar issues for several people, including me.

Death-art, 2019-10-19:
ok thank you I will try that, I will try python 3.8

ivankravets, 2019-10-19:
https://github.com/platformio/platformio-vscode-ide/issues/1207

closed 2019-10-19 by ivankravets