platformio/platformio-vscode-ide #1207: Start PIO Home Server

Death-art, 2019-10-19:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.38.1
PIO IDE: v1.9.0
System: Darwin, 16.7.0, x64

%23 Exception
```
Error: Error: Detected unknown package 'contrib-pysite'

	at S.Promise (/Users/jk/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:17383)
	at ChildProcess.h (/Users/jk/.vscode/extensions/platformio.platformio-ide-1.9.0/node_modules/platformio-node-helpers/dist/index.js:1:6150)
	at ChildProcess.emit (events.js:182:13)
	at maybeClose (internal/child_process.js:962:16)
	at Socket.ChildProcess.spawn.stream.socket.on (internal/child_process.js:381:11)
	at Socket.emit (events.js:182:13)
	at Pipe.Socket._destroy._handle.close (net.js:606:12)
```
[image]

ivankravets, 2019-10-19:
Did you delete `~/.platformio/penv` folder and restart VSCode? It seems like a problem with Python & SSL. It can't download registry manifest.

iocafe, 2019-10-19:
Hi, Psh777 has similar issue also on Darwin #1201

ivankravets, 2019-10-20:
Please provide an output from a system Terminal
```
echo $PATH
python --version
pip --version
pip list
```

ivankravets, 2019-10-24:
Could you open PlatformIO IDE Terminal, see icon on the bottom toolbar and run:
```
python -c "import requests; print(requests.get('https://api.platformio.org/stats').text)"
```

Do you see the output?

ivankravets, 2019-11-16:
Did you fix this issue?

ivankravets, 2019-11-20:
1. Please [upgrade PlatformIO IDE for VSCode to 1.10.0](https://community.platformio.org/t/platformio-ide-for-vscode-1-10/10676) or above
2. Remove `YOUR_USER_HOME_DIR/.platformio/penv` folder
3. Restart VSCode.

If you have a problem with Python interpreter, please uninstall existing [install the latest official Python 3](http://docs.platformio.org/en/latest/faq.html#install-python-interpreter).

closed 2019-11-20 by ivankravets

cnanda-zbr, 2020-06-03:
facing the exact same problem as above tried everything, doesnt seem to work.

ivankravets, 2020-06-04:
@cnanda-zbr what is your OS? Do you use antivirus tools?

cnanda-zbr, 2020-06-04:
I am using windows 10 , there is an antivirus tool but I cannot disable it.

ivankravets, 2020-06-04:
Which name of the antivirus? It seems it blocks PlatformIO. You can open PlatformIO IDE for VSCode and open it's terminal, see a terminal icon at the bottom of IDE. Type
```
pio home
```

Do you see any errors?

roetherb, 2020-06-08:
This issue is affecting everyone, I believe. The repo is explicitly blocking this package from download:

https://dl.bintray.com/platformio/dl-packages/contrib-pysite-windows_amd64-2.36.191020.tar.gz

Requesting that archive throws a 403 Forbidden.

ivankravets, 2020-06-08:
Could you try to restart VSCode? The package should be downloaded from another mirror.

cnanda-zbr, 2020-06-08:
I get the following. 
pio : The term 'pio' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was 
included, verify that the path is correct and try again.
At line:1 char:1
+ pio home
+ ~~~
    + CategoryInfo          : ObjectNotFound: (pio:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

ivankravets, 2020-06-08:
Did you click on Terminal icon located on the bottom toolbar? You can also use PlatformIO icon in the left sidebar and open PIO Core CLI. Type `pio home`.

cnanda-zbr, 2020-06-08:
when I do that i get the loading... and then nothing afterwards. 
I tried searching for a long time on other threads before arriving here finally. 
[image]

ivankravets, 2020-06-11:
What is your Internet browser where PIO Home was opened?

cnanda-zbr, 2020-06-11:
I am using the vscode console.

ivankravets, 2020-06-11:
When you type `pio home` in VSCode Console => PIO Home should be opened in your default browser. What is name/version of your browser?

cnanda-zbr, 2020-06-11:
It does not launch a new session of the browser, if that is what you are asking. My default browser is Chrome/Version 83.0.4103.97 (Official Build) (64-bit)

ivankravets, 2020-06-11:
PLEASE PLEASE PLEASE click on Terminal icon at the bottom of toolbar and run `pio home`.

cnanda-zbr, 2020-06-11:
IT IS WHAT I AM DOING  ! highlighting the picture on red for better visibility. 
[image]

ivankravets, 2020-06-11:
Do you have PlatformIO Icon in left sidebar? See
[image]

ivankravets, 2020-06-11:
Do you have PlatformIO Icon in left sidebar? See https://postimg.cc/N52yHFFj