platformio/platformio-vscode-ide #1208: Installation Manager

pcosta82, 2019-10-19:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.1
System: Windows_NT, 10.0.17763, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\pjcos\.platformio\.cache\tmp-16740e99kGdNGS2yE\virtualenv-16.7.5\virtualenv.py", line 2632, in <module>
    main()
  File "C:\Users\pjcos\.platformio\.cache\tmp-16740e99kGdNGS2yE\virtualenv-16.7.5\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\Users\pjcos\.platformio\.cache\tmp-16740e99kGdNGS2yE\virtualenv-16.7.5\virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\pjcos\.platformio\.cache\tmp-16740e99kGdNGS2yE\virtualenv-16.7.5\virtualenv.py", line 1526, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Users\pjcos\.platformio\python37\lib\shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\pjcos\\.platformio\\penv\\Scripts\\python.exe'

Using base prefix 'C:\\Users\\pjcos\\.platformio\\python37'
New python executable in C:\Users\pjcos\.platformio\penv\Scripts\python.exe

	at t.value (c:\Users\pjcos\.vscode\extensions\platformio.platformio-ide-1.9.1\node_modules\platformio-node-helpers\dist\index.js:1:32251)
```

ivankravets, 2019-10-20:
> PermissionError: [Errno 13] Permission denied

It seems line Antivirus tools block PlatformIO Installer. Try to disable for a while and restart VScode.

closed 2019-10-20 by ivankravets

pcosta82, 2019-10-20:
Hi,

Ok, I will disable the antivirus and try again.  After that, I give you
feedback.

Thank you.