platformio/platformio-vscode-ide #1210: PlatformIO Vanishing after opening Folder; Only re-appears when all folders closed.

XBrav, 2019-10-20:
IDE Version: 1.9.1
VSCode: 1.39.2
OS: 1903 64 Bit

When launching VSCode, PlatformIO will launch as expected, providing both the homescreen and the icon on the sidebar.

However, after opening any folder, the icon disappears and no PIO commands are accepted or even exposed.

See below image to confirm IDE is installed and active, but missing.

[image]

The only change to my system was the recent VS Code update to 1.39.2. I have attempted downgrading to the August builds, but the same issue persists. I have also tried re-installing the IDE.

esp32noob, 2019-10-20:
I have the same issue.  This morning platformio on VSCode worked as normal, this afternoon I can not get the platformio icons to appear.  If I open VSCode with no file open the platformio home icon appears, if I clcik on it I'm taken to the PIO Home page.  If I select a project the project opens but the platformio icon doesn't.  Here are 2 console screenshots.
Here is the console screen when VSCode opens without a project being opened:
[image]
Here is the console screen when I load a platformio project:
[image]
After reading other advice I downloaded and installed the latest version of python (3.8 64bit.)  I'm running on Windows 10.  I believe when I restarted my session that an updating message came up.  It was some time after this that I noticed the problem, i.e. I couldn't build the project.

Any advice would be gratefully received.  Thanks.

DickSwart, 2019-10-20:
Same issue here, uninstalled, reloaded, quit vscode, reopen vscode, install platformIO IDE again. Not working.

Version: 1.39.2
Commit: 6ab598523be7a800d7f3eb4d92d7ab9a66069390
Date: 2019-10-15T15:33:00.827Z
Electron: 4.2.10
Chrome: 69.0.3497.128
Node.js: 10.11.0
V8: 6.9.427.31-electron.0
OS: Darwin x64 18.7.0

```
mainThreadExtensionService.ts:65 [[object Object]]i.default.isFileSync is not a function
$onExtensionRuntimeError @ mainThreadExtensionService.ts:65
_doInvokeHandler @ rpcProtocol.ts:398
_invokeHandler @ rpcProtocol.ts:383
_receiveRequest @ rpcProtocol.ts:299
_receiveOneMessage @ rpcProtocol.ts:226
_protocol.onMessage.e @ rpcProtocol.ts:101
fire @ event.ts:580
fire @ ipc.net.ts:453
_receiveMessage @ ipc.net.ts:733
_socketDisposables.push._socketReader.onMessage.e @ ipc.net.ts:592
fire @ event.ts:580
acceptChunk @ ipc.net.ts:239
_register._socket.onData.e @ ipc.net.ts:200
t @ ipc.net.ts:28
emit @ events.js:182
addChunk @ _stream_readable.js:283
readableAddChunk @ _stream_readable.js:264
Readable.push @ _stream_readable.js:219
onStreamRead @ internal/stream_base_commons.js:94
mainThreadExtensionService.ts:66 TypeError: i.default.isFileSync is not a function
	at Object.t.isPIOProject (/Users/dickswart/.vscode/extensions/platformio.platformio-ide-1.9.1/node_modules/platformio-node-helpers/dist/index.js:1)
	at map.filter.e (/Users/dickswart/.vscode/extensions/platformio.platformio-ide-1.9.1/dist/extension.js:1)
	at Array.filter (<anonymous>)
	at c (/Users/dickswart/.vscode/extensions/platformio.platformio-ide-1.9.1/dist/extension.js:1)
	at Object.t.getActivePIOProjectDir (/Users/dickswart/.vscode/extensions/platformio.platformio-ide-1.9.1/dist/extension.js:1)
	at Object.activate (/Users/dickswart/.vscode/extensions/platformio.platformio-ide-1.9.1/dist/extension.js:1)
	at t.activate (/Users/dickswart/.vscode/extensions/platformio.platformio-ide-1.9.1/dist/extension.js:1)
	at Function._callActivateOptional (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:667)
	at Function._callActivate (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:667)
	at Promise.all.then.n (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:666)
```

neilpf2014, 2019-10-20:
Similar issue on Windows 7

Version: 1.39.2 (system setup)
Commit: 6ab598523be7a800d7f3eb4d92d7ab9a66069390
Date: 2019-10-15T15:35:18.241Z
Electron: 4.2.10
Chrome: 69.0.3497.128
Node.js: 10.11.0
V8: 6.9.427.31-electron.0
OS: Windows_NT x64 6.1.7601

from Chrome console
[Extension Host] activating extension
console.ts:137 [Extension Host] starting language server
console.ts:137 [Extension Host] (node:5124) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
t.log @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
_doInvokeHandler @ rpcProtocol.ts:398
_invokeHandler @ rpcProtocol.ts:383
_receiveRequest @ rpcProtocol.ts:299
_receiveOneMessage @ rpcProtocol.ts:226
_protocol.onMessage.e @ rpcProtocol.ts:101
fire @ event.ts:580
fire @ ipc.net.ts:453
_receiveMessage @ ipc.net.ts:733
_socketDisposables.push._socketReader.onMessage.e @ ipc.net.ts:592
fire @ event.ts:580
acceptChunk @ ipc.net.ts:239
_register._socket.onData.e @ ipc.net.ts:200
t @ ipc.net.ts:28
emit @ events.js:182
addChunk @ _stream_readable.js:283
readableAddChunk @ _stream_readable.js:264
Readable.push @ _stream_readable.js:219
onStreamRead @ internal/stream_base_commons.js:94
console.ts:137 [Extension Host] rejected promise not handled within 1 second: TypeError: i.default.isFileSync is not a function
t.log @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
_doInvokeHandler @ rpcProtocol.ts:398
_invokeHandler @ rpcProtocol.ts:383
_receiveRequest @ rpcProtocol.ts:299
_receiveOneMessage @ rpcProtocol.ts:226
_protocol.onMessage.e @ rpcProtocol.ts:101
fire @ event.ts:580
fire @ ipc.net.ts:453
_receiveMessage @ ipc.net.ts:733
_socketDisposables.push._socketReader.onMessage.e @ ipc.net.ts:592
fire @ event.ts:580
acceptChunk @ ipc.net.ts:239
_register._socket.onData.e @ ipc.net.ts:200
t @ ipc.net.ts:28
emit @ events.js:182
addChunk @ _stream_readable.js:283
readableAddChunk @ _stream_readable.js:264
Readable.push @ _stream_readable.js:219
onStreamRead @ internal/stream_base_commons.js:94
console.ts:137 [Extension Host] stack trace: TypeError: i.default.isFileSync is not a function	at Object.t.isPIOProject (c:\Users\neil\.vscode\extensions\platformio.platformio-ide-1.9.1\node_modules\platformio-node-helpers\dist\index.js:1:4689)	at map.filter.e (c:\Users\neil\.vscode\extensions\platformio.platformio-ide-1.9.1\dist\extension.js:1:13675)	at Array.filter (<anonymous>)	at c (c:\Users\neil\.vscode\extensions\platformio.platformio-ide-1.9.1\dist\extension.js:1:13658)	at Object.t.getActivePIOProjectDir (c:\Users\neil\.vscode\extensions\platformio.platformio-ide-1.9.1\dist\extension.js:1:12547)	at Object.activate (c:\Users\neil\.vscode\extensions\platformio.platformio-ide-1.9.1\dist\extension.js:1:2810)	at t.activate (c:\Users\neil\.vscode\extensions\platformio.platformio-ide-1.9.1\dist\extension.js:1:1684)	at Function._callActivateOptional (c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:667:613)	at Function._callActivate (c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:667:264)	at Promise.all.then.n (c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:666:404)
t.log @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
_doInvokeHandler @ rpcProtocol.ts:398
_invokeHandler @ rpcProtocol.ts:383
_receiveRequest @ rpcProtocol.ts:299
_receiveOneMessage @ rpcProtocol.ts:226
_protocol.onMessage.e @ rpcProtocol.ts:101
fire @ event.ts:580
fire @ ipc.net.ts:453
_receiveMessage @ ipc.net.ts:733
_socketDisposables.push._socketReader.onMessage.e @ ipc.net.ts:592
fire @ event.ts:580
acceptChunk @ ipc.net.ts:239
[... 13 lines cut ...]
_protocol.onMessage.e @ rpcProtocol.ts:101
fire @ event.ts:580
fire @ ipc.net.ts:453
_receiveMessage @ ipc.net.ts:733
_socketDisposables.push._socketReader.onMessage.e @ ipc.net.ts:592
fire @ event.ts:580
acceptChunk @ ipc.net.ts:239
_register._socket.onData.e @ ipc.net.ts:200
t @ ipc.net.ts:28
emit @ events.js:182
addChunk @ _stream_readable.js:283
readableAddChunk @ _stream_readable.js:264
Readable.push @ _stream_readable.js:219
onStreamRead @ internal/stream_base_commons.js:94
mainThreadExtensionService.ts:66 TypeError: i.default.isFileSync is not a function
	at Object.t.isPIOProject (c:\Users\neil\.vscode\extensions\platformio.platformio-ide-1.9.1\node_modules\platformio-node-helpers\dist\index.js:1:4689)
	at map.filter.e (c:\Users\neil\.vscode\extensions\platformio.platformio-ide-1.9.1\dist\extension.js:1:13675)
	at Array.filter (<anonymous>)
	at c (c:\Users\neil\.vscode\extensions\platformio.platformio-ide-1.9.1\dist\extension.js:1:13658)
	at Object.t.getActivePIOProjectDir (c:\Users\neil\.vscode\extensions\platformio.platformio-ide-1.9.1\dist\extension.js:1:12547)
	at Object.activate (c:\Users\neil\.vscode\extensions\platformio.platformio-ide-1.9.1\dist\extension.js:1:2810)
	at t.activate (c:\Users\neil\.vscode\extensions\platformio.platformio-ide-1.9.1\dist\extension.js:1:1684)
	at Function._callActivateOptional (c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:667:613)
	at Function._callActivate (c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:667:264)
	at Promise.all.then.n (c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:666:404)

neilpf2014, 2019-10-20:
Fix is to downgrade to PIO IDE extension 1.90, this is a problem with the 1.91 build.

lobocobra, 2019-10-20:
Hoi I can downgrade? This software magically updates all on its own...

sivar2311, 2019-10-20:
[image]

ivankravets, 2019-10-20:
Sorry for the issue. Please update to 1.9.2 https://github.com/platformio/platformio-vscode-ide/issues/1209

closed 2019-10-20 by ivankravets