platformio/platformio-vscode-ide #1212: Prevent automatic change of PlatformIO version installed

lobocobra, 2019-10-20:
Of course PlatformIO is a great piece of work. But some here (like me) are not professional programmer and want occasionally to change a few lines of the code added to projects like sonoff from arendst.

Instead of concentrating on the coding, it ends up always with trying to make work the environment. Of course I update on a regular basis this environment, but when I have 2h time, and want to fix a small tiny thing of my code, I do not want to end up to search for hours a solution to make my coding environment working.

**Expected behavior of PlatformIO**
PlatformIO MUST ASK everytime it wants to upgrade a library or itself. The programmer should be able to decide himself if he wants to have the newest environment OR if he just wants to focus on the work he has to do!

Thanks

nospig, 2019-10-20:
Oh yes. Just wasted a morning trying to work out why everything stopped working. The automatic update to IDE 1.9.1 broke everything and I had no clue what happened.

ivankravets, 2019-10-20:
Sorry for the issue. We did a lot of changes with 1.9 branch just to resolve a ton of issues related to Python 3 and Windows Store. We have finally switched installer to Python 3 on Windows.

If you don't want to receive updates of our extension, just go to VSCode settings and disable automatic updates.

closed 2019-10-20 by ivankravets

lobocobra, 2019-10-20:
Many thanks, I saw V1.9.2 is out and things seem to work now again.

This setting under **Settings/Application/Update.**  were not respected in older versions. I will test, if it works now and will report it.

lobocobra, 2019-10-20:
Settings of VSC are not respected.... 

I downgraded to core 3.6.7, restarted VSC and it throw an exception....

`Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: If you use Antivirus, it can block PlatformIO Installer. Try to disable it for a while.

Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\chenerkuo\.platformio\.cache\tmp-118288tFryEw6OnEW\virtualenv-16.1.0\src\virtualenv.py", line 2462, in <module>
    main()
  File "C:\Users\chenerkuo\.platformio\.cache\tmp-118288tFryEw6OnEW\virtualenv-16.1.0\src\virtualenv.py", line 762, in main
    symlink=options.symlink,
  File "C:\Users\chenerkuo\.platformio\.cache\tmp-118288tFryEw6OnEW\virtualenv-16.1.0\src\virtualenv.py", line 1015, in create_environment
    install_wheel(to_install, py_executable, search_dirs, download=download)
  File "C:\Users\chenerkuo\.platformio\.cache\tmp-118288tFryEw6OnEW\virtualenv-16.1.0\src\virtualenv.py", line 968, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
  File "C:\Users\chenerkuo\.platformio\.cache\tmp-118288tFryEw6OnEW\virtualenv-16.1.0\src\virtualenv.py", line 812, in call_subprocess
    env=env,
  File "C:\Users\chenerkuo\.platformio\python27\Lib\subprocess.py", line 390, in __init__
    errread, errwrite)
  File "C:\Users\chenerkuo\.platformio\python27\Lib\subprocess.py", line 640, in _execute_child
    startupinfo)
WindowsError: [Error 5] 

Please use the system python to run this script
Installing setuptools, pip, wheel...

  Error [Error 5]  while executing command C:\Users\chenerkuo\.platformio\penv\Scripts - setuptools pip wheel
...Installing setuptools, pip, wheel...done.

	at t.value (c:\Users\chenerkuo\.vscode\extensions\platformio.platformio-ide-1.7.1\node_modules\platformio-node-helpers\dist\index.js:1:25605)`

ivankravets, 2019-10-20:
Do you have issues with PIO Core 4?

lobocobra, 2019-10-20:
yes, it is not compatible with 3.6.7. Please see also...
https://github.com/arendst/Sonoff-Tasmota/issues/6073

With 4.0.3 I get the exception:
`c:/users/ibm_admin/.platformio/packages/toolchain-xtensa@1.40802.0/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\sonoff\firmware.elf section `.text' will not fit in region `iram1_0_seg'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\sonoff\firmware.elf] Error 1`

If I compile the same code with 3.6.7, it works perfectly.
=> When I restart VSC then PlatformIO is upgrading (without asking) and I have to delete all code folders, etc to make it work again.

ivankravets, 2019-10-20:
@lobocobra what is your platformio.ini`? It seems that you don't want to hear all of us. See again and again https://github.com/platformio/platform-espressif8266/releases/tag/v1.5.0

You need to add 
```
[platformio]
build_dir = .pioenvs
```

to your `platformio.ini`.

lobocobra, 2019-10-20:
@ivankravets 
I do read each single line of text related to the problem for the last 12 hours. Since when I platformio decided to upgrade between two starts of CSV. It took me already 2h to figure out that platformio is the problem. 

The issue opened here, is about preventing automatic upgrades.
=> I did as you suggested in Settings/Application/Update disable the automatic updates
=> To test it, I downgraded platformio core 3.6.7, compiled my code and restarted CSV
=> After restart of CSV, platformio upgrades automatically and without asking again to 4.03
In conclusion, the topic raised by me (prevent automatic and unwanted upgrade) is not yet solved.

In regard of adding build_dir = .pioenvs to platformio.ini
=> I did that of course yesterday and it did not work on 1.9.1 (as nothing was working) but it works now on 1.9.2. but as said, this is not solving the issue, that platformio is deciding itself when its upgrading. 

This means that I can not make any time capsule for code including CSV portable and until this is solved, I will need to stick with Arduino IDE.

johny-mnemonic, 2019-10-20:
> @lobocobra what is your platformio.ini`? It seems that you don't want to hear all of us. See again and again https://github.com/platformio/platform-espressif8266/releases/tag/v1.5.0
> 
> You need to add
> 
> ```
> [platformio]
> build_dir = .pioenvs
> ```
> 
> to your `platformio.ini`.

I followed here from https://github.com/arendst/Sonoff-Tasmota/issues/6073 as I also wanted just to compile my project, even without changes and I already spent more than an hour, trying to find, why without doing any single character of change I am getting such nasty error. Turns out it is auto-update of platformio :-(

So I found your post, added that build_dir reference into my platformio.ini and now my error is different:
```
Linking .pioenvs\sonoff\firmware.elf
.pioenvs\sonoff\libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'
.pioenvs\sonoff\libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'
.pioenvs\sonoff\libFrameworkArduino.a(core_esp8266_main.cpp.o): In function `loop_wrapper()':
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x21): undefined reference to `setup'
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x2d): undefined reference to `loop'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\sonoff\firmware.elf] Error 1
```
Any clue what now?

ivankravets, 2019-10-20:
> undefined reference to `setup'
> core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x2d): undefined reference to `loop'

There is no "main" file with setup/loop. Broken project?

johny-mnemonic, 2019-10-20:
@ivankravets Thanks for fast response, but no, after I gave up and downgraded to platformio 3.6.7 it compiled without any change just fine.

As I was then also able to compile with 4.x when I updated to the latest version of the project, it is clear, that there was some incompatibility between older version of the project and platformio 4.x.

So, back to issue in OP. Why platformio is updating without asking user whether he wants? Notification, with new version available and call for action (including updating later) would be much preferred!