platformio/platformio-vscode-ide #1213: Installation Manager

redlooney, 2019-10-20:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Windows_NT, 10.0.17763, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\Gus\.platformio\.cache\tmp-131728a1FlsIuAE0A\virtualenv-16.7.5\virtualenv.py", line 2632, in <module>
    main()
  File "C:\Users\Gus\.platformio\.cache\tmp-131728a1FlsIuAE0A\virtualenv-16.7.5\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\Users\Gus\.platformio\.cache\tmp-131728a1FlsIuAE0A\virtualenv-16.7.5\virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\Gus\.platformio\.cache\tmp-131728a1FlsIuAE0A\virtualenv-16.7.5\virtualenv.py", line 1526, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Users\Gus\.platformio\python37\lib\shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\Gus\\.platformio\\penv\\Scripts\\python.exe'

Using base prefix 'C:\\Users\\Gus\\.platformio\\python37'
New python executable in C:\Users\Gus\.platformio\penv\Scripts\python.exe

	at t.value (c:\Users\Gus\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:32251)
```

ivankravets, 2019-10-20:
Do you use Antivirus tools in a system?

FireDeveloper, 2019-10-20:
Machine reboot fixed it for me.

kristapsdravnieks, 2019-10-21:
I uninstalled vscode, uninstalled all python versions I had installed, rebooted my pc, and reinstalled vscode, on first launch it downloaded platformio and all worked then,

EngAdriano, 2019-10-21:
Ok.

thjubeck, 2019-12-10:
On windows disabling windows antivirus is also needed not only 3rd party antivirus solutions. Just tested for those having issues

thjubeck, 2019-12-10:
On an interesting note. For me, this was caused by a file being set with admin privileges. This was not by me so somehow the IDE set the file to admin... After copying out the file and then back in the error went away.

0x78f1935, 2020-03-02:
I removed my virtualenv folder in the project itself but my VSC settings where still pointing to the executable of python which i threw away. After selecting the system python version I could create my venv again.

ivankravets, 2020-06-03:
Hi,

Thanks for the report! We rewrote PlatformIO Core installer from the scratch. Please **download and install** the latest [PlatformIO IDE for VSCode 2.0 Beta](https://community.platformio.org/t/platformio-ide-for-vscode-2-0-beta/14097) as described in **Try Beta** section on that page. 

Please respond to this thread if it helped you or you still have an issue. We will reopen this issue again. 

Regards,
Your friends at PlatformIO

baerwa, 2020-12-03:
Hi I still have this issue. Tried to reinstall Visual Studio code + Platformio. But the issue is still there.
regards Waldemar