platformio/platformio-vscode-ide #1215: Installation Manager

kramer04, 2019-10-20:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.2
System: Windows_NT, 10.0.19002, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\dunca\.platformio\.cache\tmp-5520cbGIwWM0sKBd\virtualenv-16.7.5\virtualenv.py", line 2632, in <module>
    main()
  File "C:\Users\dunca\.platformio\.cache\tmp-5520cbGIwWM0sKBd\virtualenv-16.7.5\virtualenv.py", line 870, in main
    symlink=options.symlink,
  File "C:\Users\dunca\.platformio\.cache\tmp-5520cbGIwWM0sKBd\virtualenv-16.7.5\virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\dunca\.platformio\.cache\tmp-5520cbGIwWM0sKBd\virtualenv-16.7.5\virtualenv.py", line 1526, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Users\dunca\.platformio\python37\lib\shutil.py", line 121, in copyfile
    with open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\dunca\\.platformio\\penv\\Scripts\\python.exe'

Using base prefix 'C:\\Users\\dunca\\.platformio\\python37'
New python executable in C:\Users\dunca\.platformio\penv\Scripts\python.exe

	at t.value (c:\Users\dunca\.vscode\extensions\platformio.platformio-ide-1.9.2\node_modules\platformio-node-helpers\dist\index.js:1:32251)
```

ivankravets, 2019-10-20:
Do you use Antivirus tools in a system?

ivankravets, 2019-10-20:
@Ordsu72 Do you have antivirus tools in a system?

FireDeveloper, 2019-10-20:
Machine reboot fixed it for me.
https://github.com/platformio/platformio-vscode-ide/issues/1216

ivankravets, 2019-10-20:
See solution here https://github.com/platformio/platformio-vscode-ide/issues/1213

closed 2019-10-20 by ivankravets

kramer04, 2019-10-23:
i use windows antivirus but never said there's an issue with platformio
now it work's but sorry don't know why
i just uninstall vsc and platformio and reinstall all and take care to choose last version

Ordsu72, 2019-10-23:
> @Ordsu72 Do you have antivirus tools in a system?

Yes I have. Actually, removing Python27 & Python37 folder and copying new one worked for me. I can use Platformio again. Thank you for your help.
But I am confused a bit. Why I had two Pythons in my .platformio folder? Which one was in use?