platformio/platformio-vscode-ide #1217: Automatic installation of lib_deps before IntelliSense indexing

richard-scott, 2019-10-20:
What kind of issue is this?

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

4.0.3
### Description of problem

When opening an existing project on a new PC the lib_deps listed in the projects platformio.ini are not yet installed, this means that when you rebuild the IntelliSense Index you still get errors for not found headers etc.

I rebuild the IntelliSense index an each project opening as I often change computers, this changes my users home directory name, and this is the only way I have found to update these paths.

Anyway, it would be great to install these lib_deps prior to the IntelliSense Index rebuild to avoid these issues.

#### Steps to Reproduce

1. Create a new project and add a dependency to lib_deps in platformio.ini
2. Uninstall that library you are depending on.
3. Rebuild IntelliSense Index
4. Now go look at an "#include" line for a "file not found" issue being reported.

### Actual Results

Lib is not installed, so IntelliSense complains

### Expected Results

The lib to be installed and for IntelliSense to be OK about things.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
lib_deps =
  I2Cdevlib-ADS1115
```

**Source file to reproduce issue:**
```cpp
#include <ADS1115.h>
```

ivankravets, 2019-10-20:
We monitor folders in `lib_deps` directory. The only new folders are added, we automatically rebuild an index. Maybe, VSCode break something?

richard-scott, 2019-10-22:
I move between two PC's, I sync my code workspaces with Google Drive. If I add a new library in PC#1, and the go and work from PC#2 obviously the library isn't there, but it is referenced in my codebase and platformio.ini

Oh, and I have this in my platformio.ini
```
[platformio]
build_dir = ${sysenv.TEMP}/pio-build/$PROJECT_HASH/build
libdeps_dir = ${sysenv.TEMP}/pio-build/$PROJECT_HASH/libdeps
build_cache_dir =${sysenv.TEMP}/pio-build/$PROJECT_HASH/cache
```
This is to keep the Google Drive Sync usable, otherwise with all the file changes it gets bogged down.

I run the "Rebuild IntelliSense Index" as my username changes depending on which PC I'm on, and I've found that this rewrites c_cpp_properties.json as that needs paths updating. I have got this to trigger automatically when I open the Workspace.

I guess I'll just have to add a "build" to my automation to trigger just after rebuilding the index, but it would be really handy if both the required libs listed in the lib_deps sections (as there may be multiple build envs) of platformio.ini and the local lib_deps folder is kept in sync.

ivankravets, 2022-06-22:
Should be fixed in the latest PlatformIO Core. Please re-test.

closed 2022-06-22 by ivankravets